zephyrproject-rtos/zephyr #48098: Build error for samples/bluetooth/unicast_audio_server of nrf52dk-nrf52832 board

Fanhang32, 2022-07-21:
**Describe the bug**
Build error for samples/bluetooth/unicast_audio_server of nrf52dk-nrf52832 board

**To Reproduce**
Build with below command
`west build -p auto -b nrf52dk_nrf52832 samples/bluetooth/unicast_audio_server`

**Logs and console output**
`$ west build -p auto -b nrf52dk_nrf52832 samples/bluetooth/unicast_audio_server/
-- west build: making build dir /home/project/zephyr/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/project/zephyr/zephyr/samples/bluetooth/unicast_audio_server
-- Found Python3: /usr/local/bin/python3.8 (found suitable exact version "3.8.1") found components: Interpreter
-- Cache files will be written to: /home/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/project/zephyr/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.2 (/home/zephyr-sdk-0.14.2)
-- Found dtc: /home/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/project/zephyr/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Generated zephyr.dts: /home/project/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/project/zephyr/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/project/zephyr/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/project/zephyr/zephyr/build/zephyr/dts.cmake
Parsing /home/project/zephyr/zephyr/Kconfig
Loaded configuration '/home/project/zephyr/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration '/home/project/zephyr/zephyr/samples/bluetooth/unicast_audio_server/prj.conf'

warning: BT_ISO_PERIPHERAL (defined at subsys/bluetooth/Kconfig.iso:19) has direct dependencies (!BT_CTLR || BT_CTLR_PERIPHERAL_ISO_SUPPORT) && BT_HCI && BT with value n, but is currently being y-selected by the following symbols:
 - BT_AUDIO_UNICAST_SERVER (defined at subsys/bluetooth/host/../audio/Kconfig.baps:15), with value y, direct dependencies BT_AUDIO && BT_HCI_HOST && BT_HCI && BT (value: y), and select condition BT_AUDIO && BT_HCI_HOST && BT_HCI && BT (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /home/project/zephyr/zephyr/cmake/modules/kconfig.cmake:290 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/project/zephyr/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/project/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/project/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:91 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/.local/bin/cmake -DWEST_PYTHON=/usr/local/bin/python3.8 -B/home/project/zephyr/zephyr/build -S/home/project/zephyr/zephyr/samples/bluetooth/unicast_audio_server -GNinja -DBOARD=nrf52dk_nrf52832
`

Fanhang32, 2022-07-22:
It can build pass if I revert below patch by @rugeGerritsen,
But i don't think it's the right way to resolve this build error, who can help to give some advice? thx
https://github.com/zephyrproject-rtos/zephyr/commit/860592698be9fc401fbbf95a7610d0facdd64cc1#diff-d0914d337ff3496106eb595db0df3ff12b8f0f5004ea2efb82f119ebe381d471R21

rugeGerritsen, 2022-07-22:
@Fanhang32 , the sample is not building because of `BT_CTLR_PERIPHERAL_ISO_SUPPORT=n`, that is the Zephyr Bluetooth Controller does not support connected isochronous channels in peripheral mode yet. 

In order to get rid of this build failure you may: 
1. Not build the Bluetooth Controller as part of this sample, that is set `CONFIG_BT_CTLR=n`. This will require you to have a Bluetooth controller on another device and communicate to it using HCI. Currently it is set to `y` when enabling Bluetooth (https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52dk_nrf52832/Kconfig.defconfig#L11). 
2. Pretend that the zephyr controller supports peripheral ISO, see https://github.com/zephyrproject-rtos/zephyr/pull/46583/commits/b1ef1b0a6f90d6a2648997ae069290b07e7cdd19. Note: It will likely not work.
3. Use another Bluetooth controller

Fanhang32, 2022-07-22:
@rugeGerritsen  Thanks for your reply!
 
As i know, nrf52832 support Bluetooth 5.3, i think it should support CIS default. So is it the limitation of Zephyr Bluetooth Controller?  Do you know when CIS can be supported by Zephyr Bluetooth Controller with nrf52832?
https://www.nordicsemi.com/Products/nrf52832

rugeGerritsen, 2022-07-22:
> @rugeGerritsen Thanks for your reply!
>
> As i know, nrf52832 support Bluetooth 5.3, i think it should support CIS default. So is it the limitation of Zephyr Bluetooth Controller? Do you know when CIS [...]

A controller supporting 5.3 does not need to support all features. For controller related question, contact @cvinayak

Fanhang32, 2022-07-22:
@rugeGerritsen Got it, thanks a lot !

Hi @cvinayak , could you help to explain if nrf52832 controller can support CIS ?
If not，when can support？thanks very much