zephyrproject-rtos/zephyr #48104: [v 1.13 ] HID is not connecting to Linux based Master device

mabbas22, 2022-07-21:
**Describe the bug**
We are using nrf52840 as BLE peripheral.
The HID (peripheral_hids) of Zephyr 1.13.0 is not connecting Linux based Master device. 
I am getting following errors upon first connection request:

**_First:_**
`
ASSERTION FAIL [api->get_entropy != ((void *)0)] @
lib/zephyr/include/entropy.h:75:`

Peripheral device just reboots

**_Second_**

Security_changed callback is called and its shows: 
**_Security changed: 50:51:a9:6f:d5:1b (public) level: 2_**

After this callback the master never gets connected and i see **_0x05 authentication failed_** on connection retry

We are using **_zephyr v1.13_**, i know its very old version but most of our devices in the field are operation with it.

**Expected behavior**
Connection should be established with the master device.

**Impact**
We are currently blocked to deliver solution to our customer.

**Logs and console output**
See the attached wire_shark captures. These logs are after the security change callback called.

**Environment (please complete the following information):**

- OS: (Linux)
- Tool chain (Zephyr 1.13.0)

fabiobaltieri, 2022-07-26:
Hey, Zephyr 1.13 is no longer supported, pleas try with a more recent version.

closed 2022-07-26 by fabiobaltieri

mabbas22, 2022-08-01:
@fabiobaltieri Thanks for reply. I know that Zephyr 1.13 is not supported anymore but all of our devices in the field are based on Zephyr 1.13 and updating it recent version will take some time. We are limited to Zephyr 1.13 until we update all the devices in the field. 

Looking into the error, may be you have some ideas what is causing the issue and which commits can fix the issue for now?

Thanks a lot.