zephyrproject-rtos/zephyr #48111: License agreement not found for font software

rahulmohang, 2022-07-21:
Hello, 

There is a proprietary font software at https://github.com/zephyrproject-rtos/lvgl/blob/zephyr/examples/libs/freetype/arial.ttf. There is a reference to a license agreement. However, the location of the license agreement is not explicitly mentioned. We tried browsing to the URL of the webpage mentioned in the page. However, we could not reach the server. 

Is it possible to remove this font software?

brgl, 2022-07-22:
@kisvegabor I think this is an upstream issue, does this ring a bell for you?

kisvegabor, 2022-07-23:
Hi,

It's really an LVGL upstream issue. I replaced Arial with Lato in the  7master` branch and cherry-picked it to the `release/v8.3` branch too. I'll release v8.3.1 next week which will contain this fix.

fabiobaltieri, 2022-08-15:
This is still pending an uprev of https://github.com/zephyrproject-rtos/lvgl, @brgl can we make sure this gets addressed before the merge window closes on Sept 9th?

brgl, 2022-08-15:
@kisvegabor which commit can I cherry-pick to address this?

kisvegabor, 2022-08-15:
These two, please:
https://github.com/lvgl/lvgl/commit/7354b2ea
https://github.com/lvgl/lvgl/commit/c6f76ec7

fabiobaltieri, 2022-08-16:
Reopening until we update west.yaml to pick this up.

brgl, 2022-08-16:
@fabiobaltieri I'm on it

closed 2022-08-16 by fabiobaltieri