zephyrproject-rtos/zephyr #48147: ztest: before/after functions may run on different threads, which may cause potential issues.

smrtos, 2022-07-22:
**Describe the bug**
For the new ztest fx, there can be two threads at runtime, `main` thread and `ztest` thread.

With current implementation, the `after` function is **always** executed on the `main` thread:
https://github.com/zephyrproject-rtos/zephyr/blob/0476d6ad69f1954c5e10e48f57515b881e5a5ee1/subsys/testsuite/ztest/src/ztest_new.c#L441-L443

While the `before` function can run on `main` thread or `ztest` thread, depending on the `CONFIG_MULTITHREADING`.

On main thread:
https://github.com/zephyrproject-rtos/zephyr/blob/0476d6ad69f1954c5e10e48f57515b881e5a5ee1/subsys/testsuite/ztest/src/ztest_new.c#L434-L436

On ztest thread:
https://github.com/zephyrproject-rtos/zephyr/blob/0476d6ad69f1954c5e10e48f57515b881e5a5ee1/subsys/testsuite/ztest/src/ztest_new.c#L397-L399

These 2 threads can be in different modes (user or supervisor) and with different priority.
This may cause subtle issues.

IMHO maybe they should run on the same thread. And possibly the same thread as the test function.

**About the fix:**

With #47660, user can have a chance to run additional function-level `before/after` function on the same `ztest` thread as the test function.

Or if we don't want function-level `before/after`, we may simply move the `suite->before/after` to below function:

https://github.com/zephyrproject-rtos/zephyr/blob/0476d6ad69f1954c5e10e48f57515b881e5a5ee1/subsys/testsuite/ztest/src/ztest_new.c#L201-L206

yperess, 2022-07-26:
This is a topic I've spent a lot of time on and I don't believe should be altered. The before function runs in the same thread as the test to allow common setup to have the thread context. With that context:

Option 1: having the after function run in the same thread as before is a non starter. It results in no guarantee that the after function would ever run, making the test non-deterministic.

Option 2: having both functions run in the main test thread is possible but really just makes setup more difficult for most tests.

Option 3: having custom function level before/after is also a non starter. It doesn't solve the issue since again you'll have no guarantee that the after function will run. It would make the test non-deterministic.

Could you provide a small example of when you would need the after function to run in the same thread as the test?

smrtos, 2022-09-23:
Will update it when I have more thoughts on this.

closed 2022-12-08 by github-actions[bot]