zephyrproject-rtos/zephyr #48148: Sensor Subsystem: Base sensor DTS bindings

yperess, 2022-07-22:
We should come to an agreement on the following properties:

1, `label` should not be required, it's being phased out of Zephyr overall

2. `sensor-index` should not be specified by the application, it's assigned by the subsystem and should not be guaranteed to be the same across builds. Having to specify this manually makes it difficult to add/remove new entries in the middle.

3. `vendor`, `model`, and `friendly-name` should not be required at the individual node level, it's vulnerable to having a mismatch. Since there will be one node per sensor type here, several nodes may point to a single physical sensor.

4. I'm not sure why `minimal-interval` needs to be specified here. It's a component of the physical driver and can be attained through the sensor API.

5. `rotation-matrix` like `vendor` etc should not be here either, a physical sensor may have multiple of these nodes (1 per type) and this is likely to cause errors.

PROPOSAL

1. Remove label, there's no use for it

2. Remove `sensor-index` it should be assigned by the subsystem using `DT_` macros

3. Create a new binding for sensor metadata with all fields marked as optional. Many applications may not even have a UI and reporting the vendor/model/friendly-name is useless to them. The only required field is the sensor phandle and it should map to an existing HW sensor which we're describing.
```
properties:
  vendor:
    type: string
    required: false
  model:
    type: string
    required: false
  friendly-name:
    type: string
    required: false
  rotation-matrix:
    type: array
    required: false
  sensor:
    type: phandle
    required: true
```

4. Remove minimal interval, it's also not needed.

microbuilder, 2022-07-28:
Is it important to capture device or firmware revision at the DT level, if this isn't something that can easily be determined from registers at run time, and different device revisions behave differently in a non-trivial way?

microbuilder, 2022-07-29:
> Is it important to capture device or firmware revision at the DT level, if this isn't something that can easily be determined from registers at run time, and different device revisions behave differently in a non-trivial way?

Thinking about this, if there is value in setting revision in DT it should probably be in the specific driver binding, not the generic sensor binding since the data type may vary from one sensor to another.

yperess, 2022-08-01:
> Thinking about this, if there is value in setting revision in DT it should probably be in the specific driver binding, not the generic sensor binding since the data type may vary from one sensor to [...]

I agree, I didn't want to add more to the device API because I'm already in the process of trying to get a v2 API up and running to support more use cases (https://github.com/zephyrproject-rtos/zephyr/pull/44098). I also think that this is pretty unique to a small subset of applications that might want to expose this information to the user (the use for the vendor/model/friendly-name fields came from the HID spec). I think that having it as a separate metadata node is reasonable in order to avoid applications that don't need this information from having to compile it in.

galak, 2022-08-04:
Can you give an example dts snippet of use here.  Vendor, model info should effectively be encoded in the compatible so those don't make sense to me.  Not sure about friendly-name either.  At least from a standard zephyr binding point of view.

yperess, 2022-08-18:
> Can you give an example dts snippet of use here. Vendor, model info should effectively be encoded in the compatible so those don't make sense to me. Not sure about friendly-name either. At least from a standard zephyr binding point of view.

Sorry @galak, I didn't get a notification on this comment. Here's a sample of what I have in mind (it's crude and not 100% correct syntax), but keep in mind that I created this issue to discuss those specific properties listed in the description

```
/ {
  sensor_subsystem {
    compatible = "zephyr,sensor_subsystem";
    lid_sensor_orientation {
      compatible = "zephyr,sensor_subsystem_rotation";
      matrix = <-1 0 0
                 0 1 0
                 0 0 1>;
      apply-to = <&bmi160>;
    };
    lid_accel {
      compatible = "zephyr,sensor_subsystem_imu";
      physical_sensor = <&bmi160>;
      sensor_type = SENSOR_SUBSYSTEM_SENSOR_TYPE_ACCEL;
    };
    lid_gyro {
      compatible = "zephyr,sensor_subsystem_imu";
      physical_sensor = <&bmi160>;
      sensor_type = SENSOR_SUBSYSTEM_SENSOR_TYPE_GYRO;
    };
    base_accel {
      compatible = "zephyr,sensor_subsystem_imu";
      physical_sensor = <&bma255>;
      sensor_type = SENSOR_SUBSYSTEM_SENSOR_TYPE_ACCEL;
    };
    lid_angle {
      compatible = "zephyr,sensor_subsystem_angle";
      accelerometers  = <&lid_accel &base_accel>;
    };
};
```

Note that in this example we allocate 2 slots for sensor data that both direct requests to the same physical sensor. The virtual lid angle sensor then references these slots instead of the physical sensor. The reason is that otherwise you won't be able to arbitrate the sample rates needed for the physical sensors. They need to be wrapped once only per sensor type. This also leverages features in the Sensor V2 API #44098

huhebo, 2022-08-29:
Seems this topic is a discussion to #45370, especially for the [Device Tree Configuration](https://github.com/huhebo/zephyr/blob/subsys/sensorsubsys/doc/services/senss/index.rst#id41) section.

I do agree the proposal in #45370 is not perfect at current and need more improvements, and the #49294 is a good solution and discussion for this topic, especially for the `vendor`, `model`, `friendly-name`, `reporters`, `rotation-matrix` properties.  

@yperess, some explanations for your comments:  

> 2. sensor-index should not be specified by the application, it's assigned by the subsystem and should not be guaranteed to be the same across builds. Having to specify this manually makes it difficult to add/remove new entries in the middle.

Both chre and [windows](https://docs.microsoft.com/en-us/uwp/api/windows.devices.sensors.accelerometer.getdefault?view=winrt-22621#windows-devices-sensors-accelerometer-getdefault(windows-devices-sensors-accelerometerreadingtype)) have the `default` sensor instance concepts of a same type, such as the [chre](https://cs.android.com/android/platform/superproject/+/master:system/chre/chre_api/include/chre_api/chre/sensor.h;l=735) mentioned:  

  ```c
    For CHRE implementations that support multiple sensors of the same sensor
    type, this method can be used to get the non-default sensor(s). The default
    sensor, as defined in the chreSensorFindDefault(), will be returned if
    a sensor index of zero is specified.
  ```
  So, `sensor index` concepts is not just a internal instance id which can be generated and not consistent across builds, the `zero` index is also means the `default` sensor instance of that type, and can be get by App using `chreSensorFindDefault()`.

  I think we need make the `default` sensor instance consistent across builds, and also platform vendor need a config method to define which sensor instance is the `default` sensor.        

  Since the chre or windows only care about the `default` sensor, maybe we can just using a `default` property for the sensor instance definition in dts, but I think using a `sensor index` concepts for all sensors, and keep consistent across builds is more better.

  Or, maybe if we don't want a explicit `sensor-index` property, we can use the implicit sensor node present order in the dts as the sensor index, that's means the first present sensor instance node for a sensor type is the `default` sensor instance of this type.  

> 4. I'm not sure why minimal-interval needs to be specified here. It's a component of the physical driver and can be attained through the sensor API.

`minimal-inerval` also suitable for the virtual sensor, the consideration here is with the same sensor code implementation (virtual or physical), we can use dts to config different sensor instance with the different minimal report interval according requirements.

yperess, 2022-08-31:
> Seems this topic is a discussion to #45370, especially for the [Device Tree Configuration](https://github.com/huhebo/zephyr/blob/subsys/sensorsubsys/doc/services/senss/index.rst#id41) section.
>
> I do [...]

I think there might have been a misunderstanding with the title of this issue, my concern isn't that the sensor index exists. Just that it's specified in the device tree.

> `minimal-inerval` also suitable for the virtual sensor, the consideration here is with the same sensor code implementation (virtual or physical), we can use dts to config different sensor instance [...]

Right, but same as my comment above, why is this specified in devicetree? I'm not saying that sensors don't need to be able to report the minimal interval to the clients, just that it doesn't belong in the devicetree.

yperess, 2022-09-08:
index does not need to persistent across builds, @MaureenHelm suggested the use of chosen nodes:

```
chosen {
  zephyr,sensor-subsystem-default-accel = &lid_accel;
};
```

[chosen nodes](https://docs.zephyrproject.org/latest/build/dts/intro.html?highlight=chosen#aliases-and-chosen-nodes)

`minimal-interval` will be queried at runtime during initialization and possibly during a refresh if power state changes.

PR #50929 (2022-10-17): dts: bindings: Introduce base sensor device properties (v2)
Introduces an initial set of devicetree properties to be inherited by
all sensor devices, similar to how we define a base set of devicetree
properties for I2C and SPI devices. These properties will be used by the
future sensor subsystem to manage and expose sensors to a host operating
system, through HID or another protocol.

Fixes #48148
cc: @huhebo

This PR differs from #49294 in that it introduces only one base sensor property and migrates all in-tree sensor bindings instead of just ST and NXP. Additional base sensor properties can [... cut]