zephyrproject-rtos/zephyr #48149: Sensor Subsystem: client facing API: finding sensors

yperess, 2022-07-22:
The currently proposed API provides `senss_find_sensor()`, `senss_get_sensor_count()` which forces the caller to allocate a buffer for the sensor info structs to be filled. There's no reason to re-invent a good API and we should just use the one from CHRE.

```
int sensor_subsystem_get_sensors(const struct sensor_subsystem_sensor_info **sensors, uint16_t *num_sensors);
```

Using this model, the client doesn't need to allocate any memory, the memory is allocated exactly as needed by the subsystem (when iterating over the devicetree nodes) and we simply return a pointer to the start of the list and set the number of sensors in the list.

This leads into the `open`/`close` API. The currently proposed API takes the type & sensor_index to generate a handle which combines the two and adds a unique client identifier. This is all stored in a unit16_t (which possibly will be extended to uint32_t). Consider the following:
1. Since we have the full HID set of sensor types + extras for vendor specific ones we're already going past a uint8_t there.
2. Adding to that the index will require a few bits just to allow multiple instances of the same type.
3. In addition the unique client ID.

As a result, we'll very likely need to update this to a uint32_t instead of 16, but more so, this API makes it difficult to track client IDs across requests. An example, lets assume that the same client opens an accelerometer and a gyroscope. Using the currently proposed API, there's no way to tell that the same client is requesting both. Resources will be allocated for each independently (resources such as callback function pointers).

Instead, I would propose that we have a high level open/close API to assign clients:
```
int sensor_subsystem_open(
    const struct sensor_subsystem_sensor_callbacks *callbacks,
    sensor_subsystem_client_handle_t *handle
);

int sensor_subsystem_close(
    sensor_subsystem_client_handle_t handle
);
```

This approach would allow us to have a predetermined number of supported clients, the callbacks object provided will be set in the internal array and the handle can simply be an index into that array of the allocated client.

Subsequent calls to specific sensors can be made by a similar open/close API as the proposed one, using the combination of the sensor index from the `sensors` array populated by `sensor_subsystem_get_sensors()` and the client handle

```
int sensor_subsystem_open_sensor(
    sensor_subsystem_client_handle_t client_handle,
    uint16_t sensor_index,
    sensor_subsystem_sensor_handle_t *sensor_handle
);

int sensor_subsystem_close_sensor(
    sensor_subsystem_sensor_handle_t *sensor_handle
);
```

This will provide more fine grain resource management internally and allow us to make all the remaining calls in a similar way to the proposed API, but using the sensor_handle which encodes a single client_handle.

huhebo, 2022-08-29:
A continue discussion to #45370.

huhebo, 2022-08-29:
> The currently proposed API provides senss_find_sensor(), senss_get_sensor_count() which forces the caller to allocate a buffer for the sensor info structs to be filled. There's no reason to re-invent a good API and we should just use the one from CHRE.

> int sensor_subsystem_get_sensors(const struct sensor_subsystem_sensor_info **sensors, uint16_t *num_sensors);

> Using this model, the client doesn't need to allocate any memory, the memory is allocated exactly as needed by the subsystem (when iterating over the devicetree nodes) and we simply return a pointer to the start of the list and set the number of sensors in the list.

The problem of `sensor_subsystem_get_sensors` is it will return all sensor instances, but Apps/Clients may just want the interested sensors by type and instances index, although Apps/Clients can filter by themselves, but this will need some code and logic to do this, why we not put this common filter code and logic into sensor subsystem ? Just as #45370 did:    

  ```c
  /**
   * @brief Find sensor instances by sensor type and sensor index.
   *
   * Application can use these API to find sensor instances' information by
   * sensor type and sensor index, or enumerate all sensor instances'
   * information at one call.
   *
   * This API just returned read only information of sensor instances,
   * no side effect to sensor instances.
   *
   * @param type The sensor type which need to find,
   * \ref SENSS_SENSOR_TYPE_ALL for all types.
   *
   * @param sensor_index The index which need to find, -1 for find all
   * instances with type \p type.
   *
   * @param *info Input info array for receiving found sensor instances
   *
   * @param max The max count of the \p infos array input. Can get real count
   * number via \ref senss_get_sensor_count.
   *
   * @return Number of sensor instances foun
   */
  int senss_find_sensor(
		int type, int sensor_index,
		struct senss_sensor_info *info, int max);
  int senss_get_sensor_count(
		int type, int sensor_index);
  ```
  The caller required allocated buffer to receive the filtered sensor information, I don't think is a problem, even it's need buffer copy internally, since this API is just needed in the initial enumerate stage for a App/Client, so this also not a problem. 

  > This leads into the open/close API. The currently proposed API takes the type & sensor_index to generate a handle which combines the two and adds a unique client identifier. This is all stored in a unit16_t (which possibly will be extended to uint32_t). Consider the following:

  >  Since we have the full HID set of sensor types + extras for vendor specific ones we're already going past a uint8_t there.
    Adding to that the index will require a few bits just to allow multiple instances of the same type.
    In addition the unique client ID.

 >  As a result, we'll very likely need to update this to a uint32_t instead of 16, but more so, this API makes it difficult to track client IDs across requests. An example, lets assume that the same client opens an accelerometer and a gyroscope. Using the currently proposed API, there's no way to tell that the same client is requesting both. Resources will be allocated for each independently (resources such as callback function pointers).

We don't used the combined bits to a `uint16_t` approach (some bits for type, some bits for index, and some bits for client id etc), `typedef uint16_t senss_sensor_handle_t` is just a handle to point internal client data structure(including sensor type, index, configs etc), maybe in future, we can just use a `void *` pointer type for `senss_sensor_handle_t` directly. 

So, there is no such problem you mentioned. 

And for the `open/close` API, I don't think we need a high level `open/close` API for client level as you proposed. This makes extra step for App/Client to access sensors. 

The API design in `senss` #45370 considered various sensor App usages, including complex chre and windows HID usages, also including simple native zephyr App usages, for example, if a native zephyr App want just access a hinge angle sensor via `senss`, it can do:
 
 ```c
   /* For demo, ignored the return value checking */
   senss_sensor_handle_t hinge;
   /* Open the default hinge angle sensor (index is 0) */
   senss_open_sensor(SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE,0,&hinge);
   /* Open the second hinge angle sensor (index is 1) */
   /*senss_open_sensor(SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE,1,&hinge);*/
   senss_register_data_event_callback(hinge, data_event_callback, NULL);
   senss_set_interval(hinge, 10 * USEC_PER_MSEC);
 ```

 It's very simple and straightforward. And even for chre PAL implementation, also simple, we just enumerate all sensor instances and get a `struct senss_sensor_info` array via `senss_find_sensor()`, then create a 1:1 mapped `struct chreSensorInfo` array to  chre PAL `getSensors()` API, then we can define a 1:1 mapped `senss_sensor_handle_t` array and stored opened handles returned by `senss_open_sensor()`, all these three array using same indexing number for accessing related items. Configurations from chre PAL, then will go to related `senss` APIs with related `senss_sensor_handle_t`.     

 Although your proposal API may friendly for chre, but not for other Apps, such a native zephyr App want just access a hinge sensor sensor (please correct me if wrong): 
[... 1 lines cut ...]
```c
   /* For demo, ignored the return value checking */
[... cut ...]
   for (int i = 0; i < num, i++) {
       /* find the first instance of hinge sensor type, if App want the second hinge sensor, more logic needed here */
       if (sensors->type == SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE) {
           hinge_index = i;
	   break;
       }
   }

   sensor_subsystem_sensor_handle_t hinge;
   sensor_subsystem_open_sensor(client, hinge_index, &hinge);
   sensor_subsystem_set_interval(client, hinge_index, 10 * USEC_PER_MSEC);

 ```

 Seems it's more complex than the proposal in #45370

yperess, 2022-09-01:
> The problem of `sensor_subsystem_get_sensors` is it will return all sensor instances, but Apps/Clients may just want the interested sensors by type and instances index, although Apps/Clients can [...]

I don't believe that you'll always be able to guess what the various application filters will need to be. So I don't believe that making that filtering logic a part of the sensor subsystem is a reasonable requirement (it bloats the subsystem and makes a lot of assumptions). We will (as we both agree) have the ability to just specify that you want the `default` sensor of type X, which I think is as far as default filtering should go. Also, saying that applications can just spare the extra memory isn't reasonable, you don't know that this will only happen once as it might be possible in the future to implement plug-and-play sensors. Your requirements might also change when selecting a sensor based on other factors (like battery level etc). It's a simpler API surface to just allow the application to iterate the list, if we want we can then provide a utility that wraps that API which will provide a similar functionality to what you described, but that isn't a part of the MVP (minimum viable product).

> 
> > This leads into the open/close API. The currently proposed API takes the type & sensor_index to generate a handle which combines the two and adds a unique client identifier. This is all stored in a unit16_t (which possibly will be extended to uint32_t). Consider the following:
> 
> > Since we have the full HID set of sensor types + extras for vendor specific ones we're already going past a uint8_t there.
> > Adding to that the index will require a few bits just to allow multiple instances of the same type.
> > In addition the unique client ID.
> 
> > As a result, we'll very likely need to update this to a uint32_t instead of 16, but more so, this API makes it difficult to track client IDs across requests. An example, lets assume that the same client opens an accelerometer and a gyroscope. Using the currently proposed API, there's no way to tell that the same client is requesting both. Resources will be allocated for each independently (resources such as callback function pointers).
> 
> We don't used the combined bits to a `uint16_t` approach (some bits for type, some bits for index, and some bits for client id etc), `typedef uint16_t senss_sensor_handle_t` is just a handle to point internal client data structure(including sensor type, index, configs etc), maybe in future, we can just use a `void *` pointer type for `senss_sensor_handle_t` directly.
> 
> So, there is no such problem you mentioned.
> 
> And for the `open/close` API, I don't think we need a high level `open/close` API for client level as you proposed. This makes extra step for App/Client to access sensors.
> 
> The API design in `senss` #45370 considered various sensor App usages, including complex chre and windows HID usages, also including simple native zephyr App usages, for example, if a native zephyr App want just access a hinge angle sensor via `senss`, it can do:
> 
> ```c
>   /* For demo, ignored the return value checking */
>   senss_sensor_handle_t hinge;
>   /* Open the default hinge angle sensor (index is 0) */
>   senss_open_sensor(SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE,0,&hinge);
>   /* Open the second hinge angle sensor (index is 1) */
>   /*senss_open_sensor(SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE,1,&hinge);*/
>   senss_register_data_event_callback(hinge, data_event_callback, NULL);
>   senss_set_interval(hinge, 10 * USEC_PER_MSEC);
> ```
> 
> It's very simple and straightforward. And even for chre PAL implementation, also simple, we just enumerate all sensor instances and get a `struct senss_sensor_info` array via `senss_find_sensor()`, then create a 1:1 mapped `struct chreSensorInfo` array to chre PAL `getSensors()` API, then we can define a 1:1 mapped `senss_sensor_handle_t` array and stored opened handles returned by `senss_open_sensor()`, all these three array using same indexing number for accessing related items. Configurations from chre PAL, then will go to related `senss` APIs with related `senss_sensor_handle_t`.
> 
> Although your proposal API may friendly for chre, but not for other Apps, such as a native zephyr App want just access a hinge sensor(please correct me if wrong):
> 
> ```c
>    /* For demo, ignored the return value checking */
>    sensor_subsystem_client_handle_t client;
>    sensor_subsystem_open(event_data_callback, &client);
> 
>    struct sensor_subsystem_sensor_info *sensors;
>    int num;
>    sensor_subsystem_get_sensors(&sensors, &num);
> 
>    int hinge_index;
>    for (int i = 0; i < num, i++) {
>        /* find the first instance of hinge sensor type, if App want the second hinge sensor, more logic needed here */
>        if (sensors[i].type == SENSS_SENSOR_TYPE_MOTION_HINGE_ANGLE) {
>            hinge_index = i;
> 	   break;
>        }
>    }
> 
>    sensor_subsystem_sensor_handle_t hinge;
>    sensor_subsystem_open_sensor(client, hinge_index, &hinge);
>    sensor_subsystem_set_interval(client, hinge_index, 10 * USEC_PER_MSEC);
> ```
> 
> Seems it's more complex than the proposal in #45370

You're correct for the MVP of the subsystem this would be the right approach, though this is just to get an MVP. Additional utilities can be written around these APIs such as [chreSensorFindDefault](https://cs.android.com/android/platform/superproject/+/master:system/chre/chre_api/include/chre_api/chre/sensor.h;l=730;drc=4df2529ac0be48ac02f82b6faa1d676ba355c072).

I think the mismatch here is that I would like us to start off with as small of a product as possible. It should be fully featured but have as few APIs as possible in order to keep the design simple. Additional convenience features can be provided on top of that MVP later on.

yperess, 2022-09-02:
Discussed offline, we'll go with the CHRE API and optionally add `static inline` wrappers later on to provide filtering functionality

```
int sensor_subsystem_get_sensors(
    const struct sensor_subsystem_sensor_info **sensors,
    uint16_t *num_sensors);

static inline int get_sensors_by_type(
    sensor_type_t type,
    struct sensor_subsystem_sensor_info *sensors
    int offset,
    int max_size) {
  // call sensor_subsystem_get_sensors
  // for.. run filter
  // provide the data
}
```

closed 2022-09-02 by yperess

yperess, 2022-09-02:
As a note, the open/close aspect of this issue was not yet resolved, but that's on me. I shouldn't have mixed the two. That issue is more relevant to the data type discussion as the discussion boils down to one of the two options:
1. Combine bias and data into one event
2. Generate separate events for bias/data so they can use the same data struct

huhebo, 2022-09-02:
As we discussed offline, the major concern about `senss_find_sensor` is its has memory copy, and there are two options to resolve it: 

1. return sensor info array pointer which including all sensors, App/Client doing the filter by type and index later, this is the `sensor_subsystem_get_sensors` proposal:  
   https://github.com/zephyrproject-rtos/zephyr/issues/48149#issuecomment-1234927228  
2. return sensor info with a link-list instead of array in `senss_find_sensor`:  
   
   ```c
   int senss_find_sensor(int type, int sensor_index, sys_slist_t **sensor_info);
   ``` 
   With this approach, App/Client no need allocate memory to store sensor info, and no memory copy here, Apps/Clients get same sensor instances' constant info will point to single data structure. 

   And it's a very simple API with great flexibility, for example, App can use it with `SENSS_SENSOR_TYPE_ALL` to get all supported sensors like `sensor_subsystem_get_sensors` does, and also can use it with `SENSS_SENSOR_TYPE_MOTION_ACCELEROMETER_3D` to get only accelerometer sensors, `sensor_subsystem_get_sensors` need extra App filtering to do this.

Anyway, I think both option#1 and option#2 are all OK, I can accept option#1, but option#2 is still very worthy for consideration.

huhebo, 2022-09-02:
> As a note, the open/close aspect of this issue was not yet resolved, but that's on me. I shouldn't have mixed the two. That issue is more relevant to the data type discussion (https://github.com/zephyrproject-rtos/zephyr/issues/48150) as the discussion boils down to one of the two options:

>    Combine bias and data into one event
     Generate separate events for bias/data so they can use the same data struct

Agree,  we can continue the discussion in #48150, but I don't think we need client level open API: `sensor_subsystem_open` : 
```c
int sensor_subsystem_open(
    const struct sensor_subsystem_sensor_callbacks *callbacks,
    sensor_subsystem_client_handle_t *handle
);
```
It's introduced extra APIs and steps for Apps/Clients to open a sensor, and this API can only support single date event callback per App/Client,  but an App may want per-sensor data event callback, `senss_register_data_event_callback` provides the most flexibility, per-client (register same callback for each sensor) or per-sensor.