zephyrproject-rtos/zephyr #48175: stm32 octospi flash driver

aradenac, 2022-07-22:
**Describe the bug**
Hello, in Zephyr v3.1.0 the sample zephyr/samples/drivers/spi_flash was working fine on b_u585i_iot02a board (MX25LM51245 octospi flash), providing the following output:
[image]

But in the current main Zephyr version (commit 5432cac806b4e479a2da55afa30ead927eef3e2b) the sample program seems to be broken. This is the sample new output:
[image]

While debugging I found out that the flash driver flash_stm32_ospi.c was failing in the function stm32_ospi_config_mem at driver initialization:
[image]

Is there a known reason for this regression (a work in progress maybe) ?

galak, 2022-07-23:
This could be because of device init ordering.  Worth checking the priorities of the flash device, ospi driver, etc to make sure the init ordering is happening correctly.

fabiobaltieri, 2022-07-26:
cc @erwango

erwango, 2022-07-27:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/46740
Should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/46794.

closed 2022-07-27 by erwango