zephyrproject-rtos/zephyr #48185: LV_Z_DISPLAY_DEV_NAME symbol has not got "parent" symbol with a type

mrdimfox, 2022-07-22:
## Description

Using stm32f429i_disc1 board with Zephyr 3.1.0 I got an error on CMake configuration step. This error is connected to absent of LVGM symbols in project by default. I mean, [LV_Z_DISPLAY_DEV_NAME ](https://github.com/zephyrproject-rtos/lvgl/blob/aae1c11102138cf596687e662c0d54b72f72bf9b/zephyr/Kconfig.graphical#L12) is supposed to be declared in LVGL module with an appropriate type, but it's not, because I haven't got LVGL module in my project.

### To Reproduce

Steps to reproduce the behavior:
1. Get Zephyr 3.1.0
2. `west build -b stm32f429i_disc1 samples/hello_world`

### Expected behavior
Project is configured and built.

### Actual behavior
```
warning: LV_Z_DISPLAY_DEV_NAME (defined at boards/arm/stm32f429i_disc1/Kconfig.defconfig:17) defined without a type
Parsing /workspace/project-example/zephyr/Kconfig
Loaded configuration '/workspace/project-example/zephyr/boards/arm/stm32f429i_disc1/stm32f429i_disc1_defconfig'
Merged configuration '/workspace/project-example/zephyr/samples/hello_world/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /workspace/project-example/zephyr/cmake/modules/kconfig.cmake:290 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /workspace/project-example/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /workspace/project-example/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /workspace/project-example/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/workspace/project-example/build -S/workspace/project-example/zephyr/samples/hello_world -GNinja -DBOARD=stm32f429i_disc1 -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
```

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK 12.1.0

bbilas, 2022-07-23:
It looks similar to https://github.com/zephyrproject-rtos/zephyr/pull/48128.

mrdimfox, 2022-07-23:
> It looks similar to #48128.

Yes, as I can see these are seems related.

bbilas, 2022-07-25:
> > It looks similar to #48128.
> 
> Agreed, these are seems related.
> 
> @bbilas, could you add a stub for this symbol to your PR?

Sure, I've added `LV_Z_DISPLAY_DEV_NAME` as well. Would you mind testing that patch and confirming if it works for you?

mrdimfox, 2022-07-25:
Yeah, sure. I've tested your patch. It works as expected.

closed 2022-08-03 by fabiobaltieri