zephyrproject-rtos/zephyr #48215: docs: build the documentation failed due to "Could NOT find LATEX"

enjiamai, 2022-07-23:
**Describe the bug**
When trying to update the Zephyr documentation, there is a build error and
it will cause the PR cannot pass the CI's check of documentation.

I got a error message of  "Could NOT find LATEX" , please see https://github.com/zephyrproject-rtos/zephyr/runs/7480058629?check_suite_focus=true  for more details, which is the CI result of PR https://github.com/zephyrproject-rtos/zephyr/pull/46972.

**To Reproduce**
Steps to reproduce the behavior:
1. Push a PR which have changes of documentation.

**Expected behavior**
The documentation should pass the CI. 

**Impact**
The documentation cannot be built successfully.

**Logs and console output**
```
make[1]: Entering directory '/home/runner/work/zephyr/zephyr/doc'
cmake \
	-GNinja \
	-B_build \
	-S. \
	-DDOC_TAG=development \
	-DSPHINXOPTS="-q -W -t publish" \
	-DLATEXMKOPTS="-halt-on-error -no-shell-escape" \
	-DDT_TURBO_MODE=1
Loading Zephyr module(s) (Zephyr repository): doc
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/runner/.cache/zephyr
-- Found west (found suitable version "0.13.0a1", minimum required is "0.7.1")
-- Zephyr base: /home/runner/work/zephyr/zephyr
-- Found Doxygen: /home/runner/work/zephyr/zephyr/doxygen-1.9.4/bin/doxygen (found version "1.9.4 (5d15657a55555e6181a7830a5c7[23](https://github.com/zephyrproject-rtos/zephyr/runs/7480028816?check_suite_focus=true#step:7:24)af75e7577e2)") found components: doxygen dot 
-- Could NOT find LATEX (missing: LATEX_COMPILER PDFLATEX) 
CMake Warning at CMakeLists.txt:36 (message):
  LaTeX components not found.  PDF build will not be available.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/runner/work/zephyr/zephyr/doc/_build
cmake --build _build --target html
[0/2] Generating Devicetree bindings documentation...
[1/2] Running Sphinx HTML build...
warning: LV_Z_HOR_RES_MAX (defined at boards/arm/mimxrt1050_evk/Kconfig.defconfig:64, boards/arm/mimxrt1060_evk/Kconfig.defconfig:68, boards/arm/mimxrt1064_evk/Kconfig.defconfig:59) defined without a type
warning: LV_Z_VER_RES_MAX (defined at boards/arm/mimxrt1050_evk/Kconfig.defconfig:67, boards/arm/mimxrt1060_evk/Kconfig.defconfig:71, boards/arm/mimxrt1064_evk/Kconfig.defconfig:62) defined without a type
warning: LV_Z_DISPLAY_DEV_NAME (defined at boards/arm/stm32f4[29](https://github.com/zephyrproject-rtos/zephyr/runs/7480028816?check_suite_focus=true#step:7:30)i_disc1/Kconfig.defconfig:17) defined without a type
Warning, treated as error:
/home/runner/work/zephyr/zephyr/doc/_build/src/boards/xtensa/intel_adsp_cavs18/doc/generic_intel_adsp.rst:72:Explicit markup ends without a blank line; unexpected unindent.
FAILED: CMakeFiles/html 
cd /home/runner/work/zephyr/zephyr/doc/_build && /home/runner/.local/lib/python3.8/site-packages/cmake/data/bin/cmake -E env DOXYGEN_EXECUTABLE=/home/runner/work/zephyr/zephyr/doxygen-1.9.4/bin/doxygen DOT_EXECUTABLE=/usr/bin/dot /home/runner/.local/bin/sphinx-build -b html -c /home/runner/work/zephyr/zephyr/doc -d /home/runner/work/zephyr/zephyr/doc/_build/doctrees -w /home/runner/work/zephyr/zephyr/doc/_build/html.log -t development -q -W -t publish /home/runner/work/zephyr/zephyr/doc/_build/src /home/runner/work/zephyr/zephyr/doc/_build/html
ninja: build stopped: subcommand failed.
make[1]: *** [Makefile:20: html] Error 1
make: *** [Makefile:17: html-fast] Error 2
make[1]: Leaving directory '/home/runner/work/zephyr/zephyr/doc'
make: Leaving directory '/home/runner/work/zephyr/zephyr/doc'
Error: Process completed with exit code 2.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK, 0.14.2
 - Commit SHA: 969a3613666d706de40b4449a31dd500ac1878b1 (Not due to this commit, will provide bisect info later)

stephanosio, 2022-07-23:
> build the documentation failed due to "Could NOT find LATEX"

That is incorrect. It failed because of:

```
Warning, treated as error:
/home/runner/work/zephyr/zephyr/doc/_build/src/boards/xtensa/intel_adsp_cavs18/doc/generic_intel_adsp.rst:72:Explicit markup ends without a blank line; unexpected unindent.
```

Please fix the incorrect change in the PR.

closed 2022-07-23 by stephanosio

enjiamai, 2022-07-23:
It's my wrong. Thanks for correcting me. @stephanosio .