zephyrproject-rtos/zephyr #48216: Running gPTP sample application on SAMe54 Xplained pro , PDelay Response Receipt Timeout

Smale-12048867, 2022-07-23:
**Describe the bug**
I'm trying to run gptp demo of zephyr in the SAMe54 Xplained Pro board, 
Connected  SAMe54 Xplained Pro board and Dell Latitude 7480.
getting PDelay Response Receipt Timeout

**To Reproduce**
Steps to reproduce the behavior:
I) running the gPTP sample application on SAMe54 Xplained pro board
1. Open three terminals #1, #2 and #3
2. connect the SAMe54 Xplained pro board's debug to system
3. connect the board to system with ethernet cable (RJ45)
4. configure the board and system with IP addresses (Ex: 192.168.29.19 and 192.168.29.20)
5. Terminal#1,Build gPTP sample application for SAMe54 Xplained Pro board using west
6. Terminal#2 minicom -D /dev/ttyACM0 -o
7. Termnal#1,Flash the application using west

I) Running OpenAVNU gPTP application in terminal#3
follow https://docs.zephyrproject.org/2.6.0/samples/net/gptp/README.html

Now run Terminal#2 (Zephyr shell) run "net gptp" command.

**Expected behavior**
in terminal #3
STATUS   : GPTP [13:01:25:965] AsCapable: Enabled 

**Logs and console output**
in terminal #3
INFO     : GPTP [18:08:15:525] gPTP starting
ERROR    : GPTP [18:08:15:529] Group ptp not found, will try root (0) instead
INFO     : GPTP [18:08:15:530] priority1 = 248
INFO     : GPTP [18:08:15:530] announceReceiptTimeout: 3
INFO     : GPTP [18:08:15:530] syncReceiptTimeout: 3
INFO     : GPTP [18:08:15:530] LINKSPEED_100MB - PHY delay
			TX: 1044 | RX: 2133
INFO     : GPTP [18:08:15:530] LINKSPEED_1G - PHY delay
			TX: 184 | RX: 382
INFO     : GPTP [18:08:15:530] neighborPropDelayThresh: 800
INFO     : GPTP [18:08:15:530] syncReceiptThreshold: 8
INFO     : GPTP [18:08:15:530] SyncFollowUp with negative correction field: forbidden
ERROR    : GPTP [18:08:15:531] Using clock device: /dev/ptp0
STATUS   : GPTP [18:08:15:531] Starting PDelay
STATUS   : GPTP [18:08:15:534] Link Speed: 100000 kb/sec
STATUS   : GPTP [18:08:18:533] * Announce Timeout Expired - Becoming Master
STATUS   : GPTP [18:08:18:534] New Grandmaster "E4:B9:7A:FF:FE:4A:0B:4D" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [18:08:18:568] PDelay Response Receipt Timeout
EXCEPTION: GPTP [18:08:19:573] PDelay Response Receipt Timeout
EXCEPTION: GPTP [18:08:20:577] PDelay Response Receipt Timeout

in terminal #2
<inf> net_config: Initializing network
<inf> net_config: Waiting interface 1 (0x200009a0) to be up...
<inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
<inf> eth_sam: Link up
<inf> net_config: Interface 1 (0x200009a0) coming up
<inf> net_config: IPv4 address: 192.168.29.19
<inf> net_config: IPv6 address: 2001:db8::1                                     
uart:~$ net gptp                                                                
                                                                                
SiteSyncSync state machine variables:                                           
        Current state                  : INITIALIZING                           
        A PortSyncSync struct is ready : no                                     
ClockSlaveSync state machine variables:                                         
        Current state                  : SEND_SYNC_IND                          
        A PortSyncSync struct is ready : no                                     
        The local clock has expired    : no                                     
PortRoleSelection state machine variables:                                      
        Current state                  : ROLE_SELECTION                         
ClockMasterSyncReceive state machine variables:                                 
        Current state                  : WAITING                                
        A ClockSourceTime              : yes                                    
        The local clock has expired    : no                                     
uart:~$
**Environment (please complete the following information):**
 - OS: (ubuntu 22.04 LTS)
 - Host system : DELL latitude 7480
 - Target board: SAMe54 Xplained Pro

fabiobaltieri, 2022-07-26:
cc @nandojve

rlubos, 2022-08-09:
I don't really have this hardware, but I've tried to run the sample with `native_posix` (running vs `ptp4l`) and the sample keeps printing:
> <wrn> net_gptp: Reset Pdelay requests

@jukkar I assume it's not an expected behavior? Perhaps this and the original issue reported here are related. Both seem to be related to Pdelay message.

carlescufi, 2022-09-21:
Closing on inactivity

closed 2022-09-21 by carlescufi