zephyrproject-rtos/zephyr #48223: base64.c encode returns wrong count of output bytes

Erdyke, 2022-07-23:
**Describe the bug**
A simple off by one error in the final calculation of `olen`

**Please also mention any information which could help others to understand the problem you're facing: **
Unit testing shows amount of encoded data is different from amount expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Use base64_encode with `dlen = 0` size to get expected base64 size
2. Run base64 to encode same size of data
3. Notice `olen` is off by one.

**Expected behavior**
the final `olen` value should be calculated as `*olen = p - dst +1;`

**Impact**
Data is not copied correctly with null terminated character.

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Any

**Additional context**

Erdyke, 2022-07-23:
This might be a nevermind. Reading more carefully:
` * @note           Call this function with dlen = 0 to obtain the required buffer size in *olen`

fabiobaltieri, 2022-07-26:
@Erdyke can this be closed down then?

Erdyke, 2022-07-26:
> @Erdyke can this be closed down then?

Yes

closed 2022-07-26 by fabiobaltieri