zephyrproject-rtos/zephyr #48243: drivers: conflicting demands between DEVICE_MMIO API and GPIO drivers

ibirnbaum, 2022-07-25:
**Description:**

I was planning to refactor all device drivers for the integrated peripherals of the Xilinx Zynq-7000 / ZynqMP(UltraScale) families so that the DEVICE_MMIO API is used for mapping the device's respective register space with proper memory protection (MMU in the Zynq, MPU in the UltraScale) rather than having a DT_NODE_HAS_STATUS(..., okay)-dependent static entry for each device in the SoC init code (comp. current contents of [soc/arm/xilinx_zynq7000/xc7zxxxs/soc.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/xilinx_zynq7000/xc7zxxxs/soc.c)), as suggested in the discussion in #46463.

While the DEVICE_MMIO API works as expected in general, with the Xilinx PS UART driver being the first test subject, it now turns out that there's a conflict which prevents the supported GPIO drivers (Xilinx PS GPIO, Xilinx AXI GPIO) from being refactored:

- Each GPIO driver has in its config data struct a member of the type struct gpio_driver_config and in its run-time data struct a member of the type struct gpio_driver_data. Regarding those, the GPIO documentation states:
"This structure is common to all GPIO drivers and is expected to be the first element in the object pointed to by the config field in the device structure." (slightly different wording for the run-time data struct, but the same applies there as well).
- Each device driver using the DEVICE_MMIO_MAP API contains the macro DEVICE_MMIO_ROM in its config data struct and the macro DEVICE_MMIO_RAM in its run-time data struct, both of which resolve to the instantiation of a struct. Regarding those, the Single MMIO region macros documentation states : "This must be the first member of the config struct." ("data struct" for DEVICE_MMIO_RAM accordingly).

Following both of these demands at the same time is obviously not possible.

**Impact:**

Unless one of those two demands is more of a "should" than a "must" condition, it is currently not possible to modify GPIO drivers to use the DEVICE_MMIO API.

**Expected behavior:**

GPIO drivers can use the DEVICE_MMIO API.

**Environment:**

Current sources from main.

dcpleung, 2022-07-26:
Since both have to be the top of config/data struct, they cannot co-exist in the current foam. I can think of a couple ways forward:

1. Create another set of device MMIO API which takes a dedicated struct instead of the device struct. This means the driver will need to be aware of whether the MMIO struct resides in config or data. This is going to cause some confusions as there are two sets of device MMIO APIs.

2. Modify the current device MMIO API to not require struct at top of config/data, and simply takes in a dedicated struct. This requires modifying all current drivers using the API, and thus will require testing of them to ensure no breakage (probably not all of them, but at least more than a few).

3. Create a GPIO-specific device MMIO API in the GPIO namespace. This would co-exist with the current GPIO mandated config/data struct layout, and will only be used for device drivers requiring this.

4. Write custom code in GPIO to make it work. This is a localized version of number 3 above as this is not introducing functionality in the API level, but still need to implement it in the driver itself.

dcpleung, 2022-07-26:
@mnkp as GPIO maintainer.

carlocaione, 2022-08-08:
tagging myself. To look at this after I'm back from my vacations.

henrikbrixandersen, 2024-01-02:
> Following both of these demands at the same time is obviously not possible.

You can use named MMIO regions. See e.g. the following GPIO drivers, which use MMIO:
- `drivers/gpio/gpio_bcm2711.c`
- `drivers/gpio/gpio_davinci.c`
- `drivers/gpio/gpio_dw.c`
- `drivers/gpio/gpio_intel.c`
- `drivers/gpio/gpio_psoc6.c`
- `drivers/gpio/gpio_rcar.c`
- `drivers/gpio/gpio_sedi.c`

closed 2024-01-02 by henrikbrixandersen