zephyrproject-rtos/zephyr #48253: Only the first failing test is aborted and marked failed

alwa-nordic, 2022-07-25:
When multiple tests should fail, only the first one is aborted and marked failed. This error is present in both new and old ztest (selected by `CONFIG_ZTEST_NEW_API`).

#### How to reproduce
- Edit `tests/ztest/base/src/main.c` so that it has two or more failing tests. It works with both `ztest_test_fail();` or `zassert_true(0, NULL);` and everything else I've tried. Note that `ztest_test_fail();` is silent, so it easier to see the bad behavior with `zassert_true(0, NULL);`.
- Compile/run with `./scripts/twister --test tests/ztest/base/testing.ztest.base`.
- Look at the test output `twister-out/unit_testing/tests/ztest/base/testing.ztest.base/handler.log`.
- Notice how only the first test that should fail has "FAIL". The rest are incorrectly marked "PASS".
- If you have multiple `zassert_true(0, NULL);` in the malfunctioning test, you can see that it does not abort properly, but continues running past the failure point.
- The faulty test status is propagated to `twister-out/twister.json`, etc.

#### Output snippet
```
===================================================================
START - test_assert_tests

    Assertion failed at /home/alwa/zephyrproject/zephyr/tests/ztest/base/src/main.c:20: framework_tests_test_assert_tests: 0 is false

    Aborted at test function
 FAIL - test_assert_tests in 0.0 seconds
===================================================================
START - test_empty_test

    Assertion failed at /home/alwa/zephyrproject/zephyr/tests/ztest/base/src/main.c:14: framework_tests_test_empty_test: 0 is false

    Assertion failed at /home/alwa/zephyrproject/zephyr/tests/ztest/base/src/main.c:15: framework_tests_test_empty_test: 0 is false

 PASS - test_empty_test in 0.0 seconds
===================================================================
```

#### Version information
Zephyr commit: bdff3e76e5853fdc9dce2e72858dd9936b90d8fa (main branch)

nashif, 2022-08-02:
@yperess any idea what is going on here?

yperess, 2022-08-02:
@nashif I'm on it

closed 2022-08-10 by yperess