zephyrproject-rtos/zephyr #48287: malloc_prepare ASSERT happens when enabling newlib libc with demand paging

inteljiangwe1, 2022-07-26:
With newlibc and demand paging enabled, and config CONFIG_NEWLIB_LIBC_MIN_REQUIRED_HEAP_SIZE on demand. On malloc_prepare initization, it will trigger [assert on libc_hooks.c](https://github.com/zephyrproject-rtos/zephyr/blob/1c1e56835d8182fe45ecd0981b3738bb63cb1075/lib/libc/newlib/libc-hooks.c#L129) \
k_mem_free_get() returns vaule of available physical SRAM size, but users may actually need more. it is reasonable when enabling demand paging.

result
[image]

stephanosio, 2022-07-26:
> k_mem_free_get() returns vaule of available physical SRAM size, but users may actually need more.

So the fix would be to add more physical SRAM? This is not a Zephyr bug.

inteljiangwe1, 2022-07-26:
physical SRAM size is fixed, when enable demand paging, it will use extern memory

stephanosio, 2022-07-26:
https://github.com/zephyrproject-rtos/zephyr/blob/fb9b3bcd9326923f8c08fd6754485c28d8f6e915/include/zephyr/sys/mem_manage.h#L296-L307

> If demand paging is enabled, it may still be possible to allocate more.

@dcpleung @andyross any suggestions on how we should handle this?

andyross, 2022-07-26:
The current idea behind the demand paging design is that the app is (kinda) responsible for implementing the backing store and page replacement algorithm.  That isn't a great fit for shoehorning it in underneath the C library in a generic way.  Zephyr isn't a virtual memory OS, and fundamentally this is a request for a virtual memory process architecture.  Personally I think we should work at getting a "userspace process" abstraction worked up for C library users before doing it in demand paged virtual memory, but it's very much an open question.

I guess someone could try to do something along the lines of this request as a hack, but my gut says that will just be cruft we'd need to replace later when we try to do it right.

dcpleung, 2022-07-26:
Extending from what @andyross has said, newlib has no idea of Zephyr's memory management scheme, which means it has no idea how much memory can actually be allocated. It is possible to have a 4GB virtual memory address space, but only 256MB backing store. malloc would happily return address pointers beyond backing store capacity, and will result in system crash when the new address is accessed. That's why the current scheme is to make sure that every page that can be allocated is backed by actual physical memory, so we will not run out of space for malloc. We can possibly extend the backing store driver to tell us how much extra memory can be allocated. The mapping between virtual addresses and backing store location might be tricky if backing store contains content of zephyr.elf.

closed 2023-04-11 by github-actions[bot]