zephyrproject-rtos/zephyr #48290: ESP32 ble no work while enable CONFIG_SETTINGS

sam131208, 2022-07-26:
**Describe the bug**
 The project of zephyr\samples\subsys\mgmt\mcumgr\smp_svr, build with enable settings, will cause bluetooth loss。

**To Reproduce**
Steps to reproduce the behavior:
1. enter the project  zephyr\samples\subsys\mgmt\mcumgr\smp_svr
2. edit the prj.conf file,  enable settings:
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_MAP=y

CONFIG_SETTINGS=y
CONFIG_SETTINGS_RUNTIME=y
CONFIG_HEAP_MEM_POOL_SIZE=256
CONFIG_NVS=y
CONFIG_SETTINGS_NVS=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y

3. west build -b esp32 -- -DOVERLAY_CONFIG=overlay-bt.conf
4. west flash
5. use phone scan the esp32 ble device

**Expected behavior**
esp32 ble device can be find by lightblue or nrfconnect.

**Environment (please complete the following information):**
 - OS: (Windows)
 - Toolchain (Zephyr SDK 0.14.2)

sylvioalves, 2022-08-08:
@sam131208 This sample requires ESP32's MCUBOOT flashed prior. Have you done that?

sam131208, 2022-08-09:
@sylvioalves 
I don't know how to modify it, can you help me?

sam131208, 2022-08-09:
@sylvioalves 
I only know to modify the application's CONFIG_FLASH_INIT_PRIORITY。

sam131208, 2022-08-09:
@sylvioalves 
 In fact, another project “zephyr\samples\bluetooth\peripheral” has the same problem. it doesn't use mcuboot。

sylvioalves, 2022-08-09:
Hi, I have just tested `samples/bluetooth/peripheral` and it works as is. I can also scan the BLE over Android app.  
Can you provide more information?  

Also, which revision are you working with?  

```c
�*** Booting Zephyr OS build zephyr-v3.1.0-3244-g8fe855f166ba  ***
Bluetooth initialized
Advertising successfully started
Indicate VND attr 0x3f400668 (UUID 12345678-1234-5678-1234-56789abcdef1)
[00:00:00.568,000] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.568,000] <inf> fs_nvs: alloc wra: 0, fa0
[00:00:00.568,000] <inf> fs_nvs: data wra: 0, 0
[00:00:00.568,000] <inf> esp32_bt_adapter: BT controller compile version [1342a48]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .data initialise [0x3ffae6e0] <== [0x4000d890]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb0000] - [0x3ffb09a8]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb2730] - [0x3ffb6388]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffb8000] - [0x3ffb9a20]
[00:00:00.568,000] <dbg> esp32_bt_adapter: btdm_controller_mem_init: .bss initialise [0x3ffbdb28] - [0x3ffbdb5c]
[00:00:01.011,000] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:01.013,000] <inf> bt_hci_core: Identity: 4C:11:AE:F4:98:62 (public)
[00:00:01.013,000] <inf> bt_hci_core: HCI: version 4.2 (0x08) revision 0x030e, manufacturer 0x0060
[00:00:01.013,000] <inf> bt_hci_core: LMP: version 4.2 (0x08) subver 0x030e
[00:00:01.013,000] <inf> bt_settings: Saving ID
```

sam131208, 2022-08-09:
@sylvioalves Hi, i update the latest zephyr. samples/bluetooth/peripheral is work fine now. I noticed some updates on hal, thanks for your help。
  I will test samples\subsys\mgmt\mcumgr\smp_svr.

Thanks!

sam131208, 2022-08-10:
@sylvioalves I tested samples\subsys\mgmt\mcumgr\smp_svr again.  with the follow below feature, it is work as is.
 But, while add the CONFIG_BT_SETTINGS=y, it has the same problem and bluetooth can't work.

CONFIG_SETTINGS=y
CONFIG_SETTINGS_RUNTIME=y
CONFIG_HEAP_MEM_POOL_SIZE=256
CONFIG_NVS=y
CONFIG_SETTINGS_NVS=y
CONFIG_MPU_ALLOW_FLASH_WRITE=y

sylvioalves, 2022-08-15:
@sam131208, is this related to the #49008?

sam131208, 2022-08-16:
@sylvioalves Of course, they are all related to Bluetooth. However, I don't think they are a problem.

sylvioalves, 2022-09-21:
@sam131208 I've re-tested again enabling `CONFIG_BT_SETTINGS=y` and it still works. Can you confirm so we can close this issue?

sam131208, 2022-09-22:
@sylvioalves Thanks，i will test it.

closed 2022-09-22 by sam131208