zephyrproject-rtos/zephyr #48304: bt_disable() does not work properly on nRF52

watsug, 2022-07-26:
**Describe the bug**
With the latest Zephyr bt_disable() function was added to shutdown the BLE subsystem. It works properly on STM32WB55 and nRF53, but not on nRF52. After this sequence: bt_enable() / setup and start peripheral / bt_disable() / bt_enable() it is not possible to set up and start peripheral again. The API works, but the device is muted (not advertising again).

**To Reproduce**
Steps to reproduce the behavior:
1. bt_enable
2. create a random identity
3. setup and start advertisement using created identity
4. stop and delete the advertisement
5. delete created a random identity (the same as before)
6. bt_disable
7. bt_enable
8. reply 2-3

After step 8 there is no advertisement even if from API perspective there are no errors.

**Expected behavior**
bt_disable/bt_enable should bring BLE stack to the point where it can be used in the same way as right after the first boot.

**Impact**
Nice to have.

**Logs and console output**
Not available yet. I can add if needed.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: 2.7.2 + backports of bt_disable done on our side
 - HW: nRF52840 DK

**Additional context**
I can try to prepare a demo based on official Zephyr samples.

rugeGerritsen, 2022-08-02:
As part of fixing this issue, it would be nice if you could add a bsim test. For example doing the following:
```
for i in range 20:
    bt_enabled()
    connect()
    encrypt()
    l2cap_coc_setup()
    gatt_notify()
    bt_disable()
```

jori-nordic, 2022-11-16:
@watsug could you test again on latest master, #50046 was supposed to fix those issues.

parthitce, 2022-11-19:
> @watsug could you test again on latest master, #50046 was supposed to fix those issues.

Unfortunately this didn't help with my testing in nRF52805. The device is still not able to enable and advertise again.

jori-nordic, 2022-11-21:
@sem1-nordic could you add this procedure to the bsim testsuite and check if it completes successfully?

sem1-nordic, 2022-11-24:
Hi @watsug, I have tried to reproduce on the 52840 and seems to work fine. Can you provide any more information on your issue to try and help me see what's going wrong - an example sample that is not working, HCI logs? Is it just the nRF52805 you are having issues with now - the original ticket mentions the nRF52840 DK?

parthitce, 2022-11-24:
It's working fine in my end as well for nRF52805 with `CONFIG_SOC_FLASH_NRF_RADIO_SYNC_NONE=y` with internal RC.

carlescufi, 2022-11-24:
@sem1-nordic @jori-nordic can you please test with the Zephyr LL (as it's the only one upstream) and with different clock options (RC, Crystal, etc). Thanks!

carlescufi, 2022-11-24:
May be related to #52454

watsug, 2022-11-24:
Hello, @parthitce, I will test this ASAP and let you know.

jori-nordic, 2022-12-14:
> Hello, @jori-nordic, I will test this ASAP and let you know.

Hi, @watsug did you find the time to test it again?

jori-nordic, 2023-01-25:
Closing, feel free to re-open if it still doesn't work @watsug .

closed 2023-01-25 by jori-nordic