zephyrproject-rtos/zephyr #48334: Allow twister to naively specify additional config files and dts overlay files

yperess, 2022-07-26:
Currently, in order to specify additional `.conf` files and DTS overlays, projects need to do something like:
```
    extra_args: CONF_FILE="first.conf;second.conf" DTC_OVERLAY_FILE="first.overlay;second.overlay"
    extra_configs:
      - CONFIG_SAMPLE=y
```

I would like to add something similar to `extra_configs` for both DTC overlays and CONF_FILE so it would look more yaml native:
```
extra_configs:
  - CONFIG_SAMPLE=y
extra_conf_files:
  - first.conf
  - second.conf
extra_dtc_overlay_files:
  - first.overlay
  - second.overlay
```

yperess, 2022-07-26:
@aaronemassey @tristan-google @asemjonovs @PerMac

tristan-google, 2022-07-26:
I think this is a nice solution to a common problem we need to deal with and would make testcase.yaml files much more readable. I haven't looked closely but implementable entirely within TwisterConfigParser

nashif, 2022-07-26:
lgtm for the configs/overlays. Anyone working in this already?

Regarding parents, need some more thinking, but makes sense in general.

nashif, 2022-07-27:
> Allow twister to naively specify

I guess you meant natively? :-)

yperess, 2022-07-27:
> lgtm for the configs/overlays. Anyone working on this already?
> 
> Regarding parents, need some more thinking, but makes sense in general.

I asked @tristan-google to look at this once he has our migration to twister underway. I'm hoping he'll be able to start on it by the end of the week or early next week.

erwango, 2022-07-27:
> Long term, I would like to be able to specify a `parent` config similar to how we can specify a `common`. The parent would be a relative path and would include the parent's project fully. This would look like:
```
my.test.case:
  parent: ../common.test.case
  extra_configs:
...
```

The tricky part is to be able to make sure that, at least for .overlays, the files are taken in the expected logical order.
For instance, today,
```
common:
    timeout: 5
    extra_args: DTC_OVERLAY_FILE="foo.overlay"
tests:
  drivers.stuff_1:
    extra_args: DTC_OVERLAY_FILE="bar.overlay"
```
is different from:
```
common:
    timeout: 5
tests:
  drivers.stuff_1:
    extra_args: DTC_OVERLAY_FILE="foo.overlay;bar.overlay"
```
As the second version allows to ensure foo.overlay is applied before bar.overlay, while this is not guaranteed in the first version.
This would be nice to allow this with this new change.

tristan-google, 2022-08-03:
I will start work on the first half of this proposal (extra conf files/overlays) this week.

nashif, 2023-04-13:
done

closed 2023-04-13 by nashif