zephyrproject-rtos/zephyr #48343: NVS nvs_recover_last_ate() does not align data length

leewkb4567, 2022-07-27:
**Describe the bug**
In NVS source code 'zephyr\subsys\fs\nvs\nvs.c', nvs_recover_last_ate() walks through the ATEs. It exits the search loop when the ATE address is not larger than the data end address. The data end address is derived from the ATE found. See following extract:

`static int nvs_recover_last_ate(struct nvs_fs *fs, uint32_t *addr)`
`...`
`	while (ate_end_addr > data_end_addr) {`
`...`
`		if (nvs_ate_valid(fs, &end_ate)) {`
`...`
`			data_end_addr += end_ate.offset + end_ate.len;`
`...`

The data write is always aligned to the write block size. The data end address calculation may be made more accurate.

For nRF52 Series, the write block size is 4 bytes. If the data length is 1 byte, then there is a 3-byte gap that is not counted.

**Expected behavior**
The boundary condition may be improved by aligning the data length to the write block size.

`	data_end_addr += end_ate.offset + nvs_al_size(fs, end_ate.len);`

**Impact**
Minor

Laczen, 2022-07-27:
@leewkb4567, thank you for reporting this.

To avoid inconsistency in naming the data length (and also the data end) is always used for the 'real' data length (not the aligned). Assigning the aligned length to data end would be inconsistent and confusing. For the code it doesn't matter what end is used.

leewkb4567, 2022-07-27:
@Laczen Thanks for commenting. Agree that further aligning will not make it better, as least for nRF52.

closed 2022-07-27 by leewkb4567