zephyrproject-rtos/zephyr #48386: twister cannot take `board@revision` as platform filter

dcpleung, 2022-07-27:
**Describe the bug**
It is possible to use west build to build for different board revision, but twister does not support it.

For example, `west build -p -b nucleo_f030r8@1 samples/hello_world/` works but running `./scripts/twister -p nucleo_f030r8@1 -T samples/hello_world/` has the following error:

```
INFO    - Zephyr version: zephyr-v3.1.0-2698-gd5222f318d29
INFO    - Using 'zephyr' toolchain.
ERROR   - platform_filter - unrecognized platform - nucleo_f030r8@1
```

**Expected behavior**
Twister can using board revision in platform filter.

**Impact**
This prevents developers to test new board revisions before introducing them into the tree.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.2
 - Commit d5222f318d29bb49458b6f47a0c1f8ebed134d39

nashif, 2022-07-28:
@chen-png  Can you please take a look at this?

chen-png, 2022-07-28:
>Can you please take a look at this?

yes, I will help fix this.

dcpleung, 2022-08-09:
#48428 does not work correctly. For example, if you add revision for `qemu_x86`, `qemu_x86_64` would then be interpreted as `qemu_x86@64`.

chen-png, 2022-08-10:
>  if you add revision for `qemu_x86`, `qemu_x86_64` would then be interpreted as `qemu_x86@64`.

hi @dcpleung Could you give more details about how you reproduced this issue?
I tested to provide a qemu_x86_A.conf, it will be interpreted as qemu_x86@A, and for qemu_x86_64, it is still qemu_x86_64.

dcpleung, 2022-08-10:
If you add a dummy file `qemu_x86_64.conf`, `qemu_x86@64` would be valid due to the directory scanning code.

Another example is that, if you add `qemu_x86_tiny_1.conf`, a board `qemu_x86@tiny.1` would appear.

chen-png, 2022-08-11:
thanks, I saw your enhancement commit in https://github.com/zephyrproject-rtos/zephyr/pull/48798

dcpleung, 2022-08-18:
#48798 is merged.

PR #48428 (2022-08-09): twister: support board@revision in platform filter
support to use board@revision as platform filter when running
twister, like "twister -p nucleo_f030r8@1 ...".

met a build failure on stm32f411e_disco@B board when CI tested this PR, 
so added the fix in this PR too.

fixes: #48386
closed 2022-08-18 by dcpleung