zephyrproject-rtos/zephyr #48390: [Intel Cavs] Boot failures on low optimization levels

nklayman, 2022-07-27:
**Describe the bug**
Compiling at low optimization levels causes boot failures with both XCC and GCC toolchains across cavs platforms (tested 18 and 25, assume other platforms are affected as well). Specifically, O0 and O1 fail with XCC, while only O0 fails with GCC.

**To Reproduce**
Essentially, build and run and zephyr application on a cavs platform with O0 or O1 for XCC or O0 for GCC. Simple steps below:

1. set `CONFIG_NO_OPTIMIZATIONS=y` in hello_world prj.conf (and optionally change `compiler_flags.cmake` to force O1 instead of O0 if you'd like to test that)
2. `west build -b intel_adsp_cavs18 samples/hello_world`
3. sign binary (`west sign -t rimage --tool-path=[path to /build-rimage/rimage] -D [path to sof/rimage/config] -- -k [path to sof/keys/otc_private_key.pem]`)
4. boot with cavstool
5. see incorrect FW status and no logged message

**Expected behavior**
Hello World log message should be printed

**Impact**
Causes problems when trying to debug as I'm forced to use high optimization levels.

**Logs and console output** (from GCC with O0)
[build_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/9204073/build_log.txt)

Cavstool output:
```
INFO:cavs-fw:Use interface eno1, IP address: 192.168.68.122
INFO:cavs-fw:Ignore the interface sit0 which is not activated.
INFO:cavs-fw:Ignore the interface enp1s0 which is not activated.
/home/noah/./cavstool.py:887: DeprecationWarning: There is no current event loop
  asyncio.get_event_loop().run_until_complete(_main(None))
INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1009701)
INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
INFO:cavs-fw:Detected cAVS 1.8+ hardware
INFO:cavs-fw:Trimming 16 bytes of extended manifest
INFO:cavs-fw:Resetting HDA device
INFO:cavs-fw:Stalling and Resetting DSP cores, ADSPCS = 0xf0f0000
INFO:cavs-fw:Powering down DSP cores, ADSPCS = 0xf0f0f0f
INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
INFO:cavs-fw:type of mem is <class 'mmap.mmap'>
INFO:cavs-fw:Mapped 2M huge page at 0x46200000 to contain 90112 bytes of firmware
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:Starting DSP, ADSPCS = 0xf0f
INFO:cavs-fw:Unresetting DSP cores, ADSPCS = 0xf0f0000
INFO:cavs-fw:Running DSP cores, ADSPCS = 0xf0f0000
INFO:cavs-fw:Wait for ROM startup, ADSPCS = 0xf0f0000
INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004000
INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
INFO:cavs-fw:Waiting 2 seconds for firmware handoff, FW_STATUS = 0x5000001
WARNING:cavs-fw:Load failed?  FW_STATUS = 0x0 <-- This doesn't happen with default opt level
INFO:cavs-fw:cAVS firmware load complete
--
[This is where Hello World is printed with higher opt levels]
```

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu)
 - Zephyr SDK 14.2, XCC based off of GNU 4.2
 - Commit: 580f370867 (latest main)

**Additional context**
There's a separate issue with XCC builds on low optimization levels (#34269), but that appears to be unrelated.

marc-hb, 2022-07-28:
> Edit for clarification: The boot failure does not depend on the kconfig no_optimizations value. Hardcoding O0 into the optimize_speed causes a failure, and hardcoding O2 (or O1 for GCC) into no_optimizations results in a successful boot.

So this is making sure that there's absolutely zero CMake or Kconfiguration problem but really just an issue with the actual optimization level, and only with the optimization level correct? Did you inspect the compiler command lines with `ninja clean; ninja -v &> build.log` in all cases?

nklayman, 2022-07-30:
Yep. With XCC, I built with no_optimizations twice, with the second time hardcoding it to use O2. The build logs were identical except for the the optimization levels passed to GCC. However, only the second build successfully booted. They were also nearly identical with GCC, but there was an extra warning printed by O0.

nklayman, 2022-08-05:
Looks like this was caused by boot functions not being in the imr section, O2 would inline them so the issue wasn't noticed. There's also a weird thing with power management. See #48749 for WIP solution, turned out to be quite small (but was a lot of work to discover). Unfortunately, SOF still has its own issues (but gets further than before):
[image]

marc-hb, 2022-08-06:
@mmahadevan108 how is a failure to boot priority:low?

laurenmurphyx64, 2022-08-10:
(Update for tracking purposes) Setting up to test out Noah's PR and discussing with Flavio.

mmahadevan108, 2022-08-11:
> @mmahadevan108 how is a failure to boot priority:low?
> 
> EDIT: this seems to be an Intel-only bug BTW

This was given a priority as it was limited to certain platform.

dcpleung, 2022-08-15:
With `-O0` and `-O1`, sometimes you have to enlarge the stack sizes of various threads (e.g. main and interrupt stacks).

marc-hb, 2022-08-18:
> This was given a priority as it was limited to certain platform.

It's very high priority (and severity) for the Intel ADSP which fails to boot. Other platforms should not matter, people who don't care about Intel ADSP should filter out and never even see any intel ADSP bug anyway.

If all platform-specific bugs are automatically low priority then the priority field is  useless for platform-specific bugs.

And of course there's confusion between severity and priority as usual:
- https://github.com/zephyrproject-rtos/zephyr/issues/46759

PR #50024 (2022-10-20): intel_adsp: fix boot errors with no_optimizations
Re-opening and reworking of https://github.com/zephyrproject-rtos/zephyr/pull/48749

Fixes #48390
closed 2022-10-20 by nashif