zephyrproject-rtos/zephyr #48414: Make it easier to find release dates and lifetimes of releases

Thalley, 2022-07-28:
**Is your enhancement proposal related to a problem? Please describe.**
The release dates of Zephyr is mainly only visible from https://github.com/zephyrproject-rtos/zephyr/releases, but the search feature is quite bad and it's hard (or at least harder) to find the release you are looking for a release date from. 

**Describe the solution you'd like**
Something like the release table from https://www.python.org/downloads/ would be nice, where you have the release date and the end of life support. This could be expanded with information about LTS and if there are "subreleases" like 2.7.1 and 2.7.2. 

This information should also be available on docs.zephyrproject.org/ and not Github. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

henrikbrixandersen, 2022-07-28:
Perhaps we could add "(LTS)" to the LTS releases on https://docs.zephyrproject.org/latest/releases/index.html

Thalley, 2022-07-28:
> Perhaps we could add "(LTS)" to the LTS releases on https://docs.zephyrproject.org/latest/releases/index.html and list their EOL in the respective release notes?

That would also be nice. The list of releases we have is nice, but adding EOL and date on those would make it very easy to get an overview

chrta, 2022-07-29:
Another example for a release overview: https://wiki.yoctoproject.org/wiki/Releases

keith-zephyr, 2023-10-17:
@kartben -  Is this something you could look into?  I'm not sure this warrants discussion by the Process WG.

nashif, 2023-10-18:
IMO this is already done here https://docs.zephyrproject.org/latest/releases/index.html#supported-releases and it is not a process related issue