zephyrproject-rtos/zephyr #48463: Grant Triage permission level to @aurel32

erwango, 2022-07-29:
# Nomination

Grant Triage permission level to @aurel32

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Aurelien Jarno <aurelien@aurel32.net>
1. GitHub username: https://github.com/aurel32

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pulls?q=is%3Apr+author%3Aaurel32+

erwango, 2022-07-29:
@aurel32

stephanosio, 2022-07-29:
@aurel32 already has triage permission.

stephanosio, 2022-07-29:
@aurel32 I am actually surprised that you are not in the MAINTAINERS file given the amount of contributions you have made.

Could you open a PR adding yourself as a collaborator for one of the areas you have been involved in so that we can add you as a collaborator?

erwango, 2022-07-29:
> @aurel32 I am actually surprised that you are not in the MAINTAINERS file given the amount of contributions you have made.
> 
> Could you open a PR adding yourself as a collaborator for one of the areas you have been involved in so that we can add you as a collaborator?

Ok, let me close this issue and we'll check this.

closed 2022-07-29 by erwango