zephyrproject-rtos/zephyr #48511: STM32WL LoRaWAN pinout incomplete / RFO pin not selectable

senseDan, 2022-08-01:
**Is your enhancement proposal related to a problem? Please describe.**

The STM32WL chip series has on chip LoRa and provides two input and two output pins for this. The low power output pin (RFO_LP) and the high power output pin (RFO_HP) are currently not choosable. By default the RFO_HP pin is used.

The limitation was found because of the following problem:
A custom board with STM32WLE5CC, which is optimized to only use the RFO_LP pin was configured with two different firmwares.
The first firmware is a bare metal version configured with STM32CubeMX and the second one is the lorawan class a sample from zephyr. The bare metal firmware reached and rssi of around -70 and the zephyr sample only of -110. After connecting the RFO_LP and the RFO_HP pin the zephyr sample reached an rssi of around -60.

**Describe the solution you'd like**
An optimal solution would be an option in the devicetree to select the preferred RFO pin.

**Additional context**
From the Datasheet of STM32WLE5CC
[image]

erwango, 2022-08-05:
Rather looks like a pure LORA driver point.

aurel32, 2022-08-05:
> An optimal solution would be an option in the device tree to select the preferred RFO pin.

The SX127X driver has such an option, but not the SX126X driver, as both the SX1261 and SX1262 chips only have a single RF output, respectively LP and HP. The HP and LP output seems specific to the ST implementation, and the way to switch is not documented in the STM32WL reference manual, nor in the SX126X manual.

That said it is possible to find the way to switch between the two implementation by looking at STM32Cube. It appears that it uses a modified version of the loramac-node library, and uses the bit that selects between the SX1261 and the SX1262. So you might be able to get the LP output working by changing SX1262 into SX1261 in these lines:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/lora/sx126x_common.h#L25-L27

> Rather looks like a pure LORA driver point.

If that works it should be possible to add a property in the `st,stm32wl-subghz-radio` binding and use it to select the correct chip version.

erwango, 2022-08-05:
> The HP and LP output seems specific to the ST implementation, and the way to switch is not documented in the STM32WL reference manual,

IIUC, it's actually documented in §5.8.4:

> Set_PaConfig() command:
byte 3
    bits 7:1 Reserved, must be kept at reset value.
    bit 0 PaSel: PA selection.
      0: HP PA selected
      1: LP PA selected (default)

aurel32, 2022-08-05:
> IIUC, it's actually documented in §5.8.4:

Indeed I search too quickly through the reference manual. Apologies.

erwango, 2022-08-05:
> > IIUC, it's actually documented in §5.8.4:
> 
> Indeed I search too quickly through the reference manual. Apologies.

Took me some time to find it :-D

senseDan, 2022-08-08:
Thanks, so I am adding this up from you.
According to the STM32WLE5CC RM (https://www.st.com/resource/en/reference_manual/rm0461-stm32wlex-advanced-armbased-32bit-mcus-with-subghz-radio-solution-stmicroelectronics.pdf) in §4.8.4 the Set_PaConfig(PaDutyCycle, HpMax, PaSel, 0x01) sets to LP or HP RFO.

In `modules/lib/loramac-node/src/radio/sx126x/sx126x.c` the function `void SX126xSetTxParams( int8_t power, RadioRampTimes_t rampTime) ` calls the set_PaConfig() function according to the device. This is for SX1261 a LP setting and for SX1262 a HP setting.

> That said it is possible to find the way to switch between the two implementation by looking at STM32Cube. It appears that it uses a modified version of the loramac-node library, and uses the bit [...]
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/lora/sx126x_common.h#L25-L27

According to these lines for STM32WL the SX1262 is selected and so HP output is selected. 

So I just changed line 27 from `#define SX126X_DEVICE_ID SX1262` to `#define SX126X_DEVICE_ID SX1261` and it seems to work. My test with the rssi (of course the RFO pins in this case NOT connected) improved from -110 to -70.

danielSchaeble, 2023-02-23:
An idea for a solution in the zephyrproject discord:

https://discord.com/channels/720317445772017664/883448180950327366/1072593303729623100

PR #56833 (2023-04-24): lora: Add support for fine-grained STM32WL power amplifier configuration
### Summary

This PR adds the ability to select between the low-power `RFO_LP` and high-power `RFO_HP` LoRa radio power amplifier output, as well as setting the maximum design Tx output via a new devicetree properties on STM32WL-based boards.
This allows better matching of the power amplifier output with boards' RF impedance matching network.

Previously only the high-power `RFO_HP` output was supported in the highest design power output (+22 dBm), as the same PA config register is used to differentiate between the discre [... cut]
commit e78e8d7c (2023-04-24): drivers: lora: sx126x: Support fine-grained STM32WL PA config

Add STM32WL-specific sx126x_set_tx_params function based on the
STM32CubeWL modifications to LoRaMac-node.

Add the "power-amplifier-output" DT property to
"st,stm32wl-subghz-radio" for selecting between the RFO_LP and RFO_HP
output configurations provided by the above mentioned function.

Add the "rfo-lp-max-power" and "rfo-hp-max-power" DT properties for
defining the maximum design power of the respective outputs' matching
networks.

closes #48511
closed 2023-04-24 by carlescufi