zephyrproject-rtos/zephyr #48518: samples/sensor/*: Build issue when board expose sensors defined on both I2C and SPI buses

erwango, 2022-08-01:
**Describe the bug**
There are a few sensor samples which fail to build with `96b_argonkey`:
`samples/sensor/vl53l0x/`
`samples/sensor/hts221`
`samples/sensor/lps22hb`

```
/<redacted>/zephyr-sdk/sdk-0.14.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/sensor/lsm6dsl/libdrivers__sensor__lsm6dsl.a(lsm6dsl.c.obj):(.rodata.lsm6dsl_config_0+0x4): undefined reference to `__device_dts_ord_83'
```

Root cause is:
- Board dts defines `lsm6dsl` on SPI bus, while other sensors (`hts221`, `vl53l0x`, `lps22hb`) are defined on I2C bus
- samples for I2C sensors enable both `CONFIG_SENSOR` and `CONFIG_I2C`
- Due to `CONFIG_SENSOR=y` `drivers/sensor/lsm6dsl.c` is compiled, but compilation fails due to `CONFIG_SPI` not being enabled.

This is likely an effect of:
```
menuconfig LSM6DSL
	bool "LSM6DSL I2C/SPI accelerometer and gyroscope Chip"
	default y
+	depends on DT_HAS_ST_LSM6DSL_ENABLED
	depends on I2C || SPI
```

Ideally `LSM6DSL` activation should be a little bit more selective and check bus activation:
```
	depends on DT_HAS_ST_LSM6DSL_ENABLED
	depends on ((I2C && DT_LSM6DSL_ENABLED_ON_I2C_BUS) || 
                            (SPI && DT_LSM6DSL_ENABLED_ON_SPI_BUS))
```

galak, 2022-08-02:
@erwango I had worked up something similar to what you suggested in a PR:

https://github.com/zephyrproject-rtos/zephyr/pull/48048

```
depends on (I2C && $(dt_compat_on_bus,$(DT_COMPAT_ADI_ADXL345),i2c)) || \
		   (SPI && $(dt_compat_on_bus,$(DT_COMPAT_ADI_ADXL345),spi))
```

However @MaureenHelm asked to just update the board Kconfig.defconfig to enable the bus needed.  So will look for her comment on this to see what direction we should go.

erwango, 2022-08-02:
> @MaureenHelm asked to just update the board Kconfig.defconfig to enable the bus needed

If this is something like follows, this would make sense indeed:
```
config SPI
    default y if LSM6DSL

config I2C
   default y if VL53L0X || HTS221 || LPS22HB
```

MaureenHelm, 2022-08-02:
> ```
> depends on (I2C && $(dt_compat_on_bus,$(DT_COMPAT_ADI_ADXL345),i2c)) || \
> 		   (SPI && $(dt_compat_on_bus,$(DT_COMPAT_ADI_ADXL345),spi))
> ```

I really don't like this pattern.

> If this is something like follows, this would make sense indeed:
> 
> ```
> config SPI
>     default y if LSM6DSL
> 
> config I2C
>    default y if VL53L0X || HTS221 || LPS22HB
> ```
> 

I tried this before, but it created a circular dependency (LSM6DSL depends on SPI, and SPI depends on LSM6DSL). That's why I wound up using SENSOR instead of LSM6DSL.

```
config SPI
	default y if SENSOR
```

> Edit: There would be a consequence on `samples/sensors/*/prj.conf` since enabling SENSOR would then automatically bring up buses. So `CONFIG_I2C=y` and `CONFIG_SPI=y` would have to be cleaned up. Another consequence is that in case of board has sensor on both bus. Enabling SENSOR would bring up both buses automatically. But this could be fine tuned using .overlays, so that is not a concern to me.
> 

Agreed

galak, 2022-08-04:
@erwango will you work up a PR for this?

erwango, 2022-08-04:
> That's why I wound up using SENSOR instead of LSM6DSL.
```
config SPI
	default y if SENSOR
```

I'm not convinced by this solution. This enables bus irrespective of sensor devicetree status on the bus.
We end up in a situation where:
- If I have sensors on one type of bus, bus is activated only if required based on devices device tree status.
- If I have sensors splits on 2 types of bus, both bus are activated irrespective of devicetree status (as long as one sensor is enabled). If I want to enable sensors on one type of bus, I need to update device tree and manually edit `board_config` to disable the bus not needed.

So I kind of prefer solution proposed [here](https://github.com/zephyrproject-rtos/zephyr/issues/48518#issuecomment-1202422501)

erwango, 2022-08-04:
Dev meeting: Review the bus selection: select bus in drivers and remove board related code.
I'll have a try and propose a PR

galak, 2022-08-04:
> I'll have a try and propose a PR

pushed a PR, will see how it goes through CI.

commit df81fef9 (2022-08-08): drivers: sensor: Convert Kconfig bus 'depends on' to 'select'

This change in pattern is meant to address a misconfiguration issue
that can occur for sensors that support being on multiple busses
like I2C & SPI.

For example, you can have a configuration in which such a sensor is
on the I2C bus in the devicetree and the sensor is enabled.  However
the application configuration enables CONFIG_SPI=y and CONFIG_I2C=n
and this will cause the sensor driver to be built by default, however
since we don't have the I2C bus enabled the driver will not compile
correctly.

Previously we had been adding to [... cut]
closed 2022-08-08 by fabiobaltieri