zephyrproject-rtos/zephyr #48520: clang-format: #include reorder due to defalut: SortIncludesOptions != SI_Never

ppryga, 2022-08-01:
**Describe the bug**
If use `clang-format` to apply formatting on save and e.g. add #include<xxx.h> in a source file then it runs sorting of all #includes.

The problem is caused by change of `.clang-format` to use `BasedOnStyle: LLVM`. This style, by default, has enabled `SortIncludes` that causes an issue. This option has to be disabled, in other case it makes a mess with `#include` directives.
Unfortunately the `SortIncludes` has been changed from SortIncludes (bool) to `SortIncludes (SortIncludesOptions)`. So we have to stick to some recent version of a clang-format causing lack of a backward compatibility.

**Expected behavior**
#include directives are not changed by formatting tool.

**Impact**
Breaks compilation due to internal dependencies between header files.

gmarull, 2022-08-11:
If sorting includes introduces issues, headers should be fixed to be self-contained instead.

ppryga, 2022-08-12:
Well that may be difficult.... There are dependencies between headers. I'm not sure it making them standalone is possible at all. Even if it is that would be significant effort to satisfy tool. In my humble opinion the tool has to satisfy us.

gmarull, 2022-08-12:
> Well that may be difficult.... There are dependencies between headers. I'm not sure if making them standalone is possible at all. Even if it is that would be significant effort to satisfy tool. In my humble opinion the tool has to satisfy us.

It should be possible, non-self contained headers should be treated as a bug IMO. 

https://google.github.io/styleguide/cppguide.html#Self_contained_Headers

nashif, 2022-08-12:
> It should be possible, non-self contained headers should be treated as a bug IMO.

ok, but we are not there yet and IMO this should not be part of the default style configuration we use, maybe  once all issues are resolved we could enable that, right now it is just introducing potential issues for anyone trying to use clang-format for the rest of guidelines.

gmarull, 2022-08-12:
> ok, but we are not there yet and IMO this should not be part of the default style configuration we use, maybe once all issues are resolved we could enable that, right now it is just introducing [...]

Certain areas will take a loooong time to get fixed (if they ever do because if things _work_ there's no real incentive to improve unless it is enforced somehow). I think that areas doing things well should be using this option to avoid yet another file reformat in the future. When working with buggy headers, one can disable that option locally, or, resolve the problem.

ppryga-nordic, 2022-08-15:
> Certain areas will take a loooong time to get fixed (if they ever do because if things _work_ there's no real incentive to improve unless it is enforced somehow). I think that areas doing things [...]

That is a bit uncomfortable for all maintainers that changes something in the areas where headers are no-self contained. Enable-disable the feature depending what area is updated. I suggest to disable it globally and enable it when we are actually ready for that step. I support @nashif opinion here.

ppryga-nordic, 2022-10-17:
PR provided by @nashif #49293 has two approvals and can be merged. Then the issue may be closed.

PR #49293 (2022-11-16): clang-format: disable regrouping and sorting of includes
Disable regrouping and sorting of includes which can cause issues due to
order requirements in some subsystems.

Fixes #48520
closed 2022-11-16 by carlescufi