zephyrproject-rtos/zephyr #48531: RFC: Changing the sys_clock interface to fix race conditions.

peter-mitsis, 2022-08-01:
## Introduction

Recently https://github.com/zephyrproject-rtos/zephyr/pull/48474 was opened to address a race condition in sys_clock_announce(). Additional issues in the existing implementation have since been identified and this RFC is an attempt to both accurately describe them and (hopefully) get agreement about the solution proposed here. It affects every timer driver and the sys_clock_announce() kernel routine.

### Problem description

In short, the current design generally has the timer ISR save the cycle count at which the last tick occurred (last_count) for use later in sys_clock_set_timeout(). As we currently use two different spinlocks (a timer specific one and a separate timeout_lock spinlock in timeout.c) there is a window of opportunity where *last_count* can be changed which in turn affects when the next timer ISR is serviced.

### Proposed change

I propose that we get rid of the two-spinlock implementation and simply use the kernel's timeout_lock spinlock in both the timer driver and timeout.c. This necessitates that the following 10000 foot changes.
1. Timer ISR locks the timeout_lock spinlock upon entry.
2. Timer ISR passes the key to the spinlock to sys_clock_announce()--this is an API change.
3. sys_clock_set_timeout() assumes that it is called with the timeout_lock spinlock already acquired so it does not need to acquire it.
4. sys_clock_announce() releases the timeout_lock spinlock before it returns.

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

Here's how the structure of a tick delivery would look.

```
timer_isr()
{
Acquire the timeout_lock spinlock
Determine number of elapsed ticks
Update when last system clock tick occurred
sys_clock_announce(elapsed ticks, key to spinlock)
/* spinlock is released by sys_clock_announce */
}

```

```
/* timeout_lock spinlock is already acquired */
sys_clock_announce(delta ticks, key to spinlock)
{
Do timeslice work.

/* Improvement: This early bailout check applies to both UP and SMP */
If ((timeout list being processed) or (delta ticks is 0)) {
    Update announce_remaining ticks
    Release timeout_lock spinlock
    return;
}

/* No change to this loop outside of what is proposed in PR #48474 */
Process timeout list {
    Update both current and announce_remaining ticks
    Release timeout_lock spinlock
    Invoke callback
    Re-acquire timeout_lock spinlock
}

Indicate we are done processing timeout list
Update both current and announce_remaining ticks
Call sys_clock_set_timeout(next timeout, false)

Release timeout_lock_spinlock     **<--- This is new**
}

```

```
/* timeout_lock spinlock is assumed to be locked upon entry*/
sys_clock_set_timeout()
{
Unchanged except that it no longer acquires/releases the old driver spinlock
This references the time at which the last system clock tick occurred.
}
```

### Dependencies

As I am proposing that a new parameter be passed to sys_clock_announce(), this impacts every timer driver's timer ISR as they call sys_clock_announce().

### Concerns and Unresolved Questions

I think this covers all the race conditions that may arrive from an SMP system, but it is possible that I am missing something. 

## Additional Benefits

1. In an UP system, we reduce the time spent in a nested timer interrupt on a heavily loaded system since a nested timer interrupt will bail out early.
2. In an SMP system, I am confident that we ensure that only the time at which the last system clock tick occurred will not change underneath us.
3. Although every timer driver is affected, the changes are minimal.

andyross, 2022-08-02:
The immediate problem this will cause (really regress[1]) is that simultaneous timer ISRs will now block each other.  Remember that due to timeslicing requirements[2], we require timer interrupts be delivered to all cores.  The current architecture allows a timer driver to compute a tick count to announce, enter the announce call, and then early-exit knowing that another CPU is going to handle it.  With this scheme all the cores will block in unison, which isn't really desirable.

Also: as I see it there's no particular race inherent in the existing design.  The computation of elapsed ticks can be reliably done atomically, and the announcement can be done in a race-free manner.  That there is a plausible misordering bug is something that can just be fixed, right?

[1] The core issue that led to the discovery of the misordered timeout handler thing a few months ago wasn't that (which was never seen), it was a performance bug in intel_adsp where we'd slip ticks due to the colliding handlers blocking the world.

[2] And maybe that's the architecture we really want to be pointing at instead.  Have the per-CPU timeslice events be regular k_timeout objects.  Add a cpu-targetted IPI variant (which can be implemented via the existing broadcast one for existing platforms as a compatibility fallback).  And just poke the needed CPU when it gets interrupts.  Then we can move back to an architecture where the timer ISR needs to be handled on only one CPU and avoid SMP worries (though there would still have to be some protection for correctness reasons, but it could be a big global lock as working drivers wouldn't collide).

andyross, 2022-08-02:
Per a comment at the dev meeting: maybe it would help to write up an event sequence that shows the race in the current setup?  That is, a list of what happens when that ends up at an incorrect action (in this case, presumably a timeout failing to fire early enough, or a handler failing to execute, or executing at the wrong time, I guess?).

andyross, 2022-08-02:
Oh, and also per the meeting, @carlescufi  pointed out this bit I wrote was ambiguous:

> Also: as I see it there's no particular race inherent in the existing design. The computation of elapsed ticks can be reliably done atomically, and the announcement can be done in a race-free manner. That there is a plausible misordering bug is something that can just be fixed, right?

The point is that the existing misordering bug (IIRC: the last timeout entry in the list can end up executing in parallel with a newly-added-just-now timeout) can be fixed within the existing timeout subsystem lock.  It doesn't require coordination with the other lock in the timer driver.

peter-mitsis, 2022-08-02:
The scenario I was thinking of goes like this ...

Core 0 starts servicing the timer ISR. It does the ISR work (acquire driver spinlock, set last_count, calculate elapsed ticks, and unlocks the driver spinlock).
Core 0 invokes sys_clock_announce().
Core 0 processes the timeout list.
Core 0 calls next_timeout() to determine when the next timer ISR should fire.
Core 1 finally has time to process the timer interrupt. It gets the driver spinlock.
Core 0 calls sys_clock_set_timeout() and blocks on the driver spinlock.
Core 1 updates last_count and calculates that N ticks have elapsed (N > 0)
Core 1 releases the driver spinlock and calls sys_clock_announce.
Core 0 gets the driver spinlock and programs the timer based on Core 1's updated last_count. However, this is N ticks too far into the future from what core 0 previously calculated.

andyross, 2022-08-02:
> Core 0 gets the driver spinlock and programs the timer based on Core 1's updated last_count. However, this is N ticks too far into the future from what core 0 previously calculated.

Maybe elaborate this to explain why the resulting app behavior becomes incorrect?  I content this is correct behavior.  "Time" is an aspect of the physical universe, it can't be "locked" and is not subject to synchronization.  We don't want core 0 running "in the past", we want timeouts that it sets to reflect the actual time.  (There's some complexity here for sure, because in fact the system time does indeed "freeze" during the execution of a timeout handler, such that the handler gets to assume it's running at the right time and we don't have racing contexts seeing seemingly different timelines.  But when it comes back it comes back to real time.)

peter-mitsis, 2022-08-02:
I’m playing this scenario out more, and I am starting to think that I had previously missed something because I stopped too early. Here’s my thinking ....

If in the previously described scenario core 1 has ticks to announce, then it is true that core 0 will set the timer to expire later in the future than expected. This is where I had previously stopped in my thought experiment as I was expecting consistent states all the way through. However, progressing forward in the thought experiment ... core 1 executes and processes any items in the timeout list. If anything was processed, this would include the first item which would have impacted the value to which core 0 set the timer, so it is still going to get serviced on time.  Ultimately core 1 will set the timer to correctly expire in some number of ticks. If we do not have a repeat of the scenario, then this will be the correct  value for the timer. If the scenario repeats with a different set of cores, then  stuff still gets processed in order until the chain is broken and the timer is ultimately programmed with the correct value.

If the above is correct, then the major concerns I had are gone.  However, we might still be able to have an extra "bail out" condition in sys_clock_announce() for cases when *ticks* is zero. The thinking being that if there are no ticks to announce, then the there will be no change to the timeout list nor to when the timer interrupt should fire. (However, I will need to think a little more on that as it is evident that I missed some logic already.)

peter-mitsis, 2022-08-04:
The bug to which we were trying to solve has been resolved without the need to go the route of changing the sys_clock_xxx interfaces. Consequently, I am closing this item.

closed 2022-08-04 by peter-mitsis