zephyrproject-rtos/zephyr #48534: SMF missing events

nRF5xFanBoy, 2022-08-01:
**Is your enhancement proposal related to a problem? Please describe.**
The problem is with the current state machine framework is I can not de-queue events in the current state!
Also missing examples in the zephyr folder examples.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I wish for state aware de-queue dispatching (due to a missing example in the zephyr folder examples).
I do wish same features as with very powerful state-machine compiler [http://smc.sourceforge.net/]

**Describe alternatives you've considered**
Currently I need very unwanted over-complex states with run & exit routines to dispatch specific events.

**Additional context**
Please have a sneak peak at the powerful state machine compiler for generated code of the SMC
http://smc.sourceforge.net/

sambhurst, 2022-08-03:
I'm unsure what you mean about de-queuing an event in the current state. Could you explain further and maybe add some sudo-code? Also, the [documentation](https://docs.zephyrproject.org/latest/services/smf/index.html) has two complete examples that can be extracted and built.

Personally, I'm not a big fan of state machine compilers because you're dealing with two languages instead of one and you still need to flesh out the states after they've been generated.

nRF5xFanBoy, 2022-08-13:
@sambhurst 

> "Personally, I'm not a big fan of state machine compilers because you're dealing with two languages instead of one and you still need to flesh out the states after they've been generated."

How can I create a event drive firmware if states can't handle incoming events?

> "Also, the [documentation](https://docs.zephyrproject.org/latest/services/smf/index.html) has two complete examples that can be extracted and built."

This SMF is just about states, it can't handle incoming events? Can you please extend the examples to a event driven example.

> "because you're dealing with two languages instead of one"

Generated code hides (abstraction) the boring complex code, which takes days for a human to type (so it is a time safer and less erroneous.
A second language is less a problem if that is a higher level language more closer to human language. 
The SMC (http://smc.sourceforge.net/) is very easy to understand.

nRF5xFanBoy, 2022-08-13:
@sambhurst
In my humble opinion I do think that zephyr lack some basic building blocks to support an event-driven firmware application. Correct me please, perhaps I did overlook some documentation and/or samples?

sambhurst, 2022-08-15:
@nRF5xFanBoy
You can create an event driven state machine by blocking on or polling for an event and passing it to the smf_run_state state machine API and this will pass the event to the current state, where it can be acted upon. Something like the following:

```
/* Run the state machine */
while(1) {
     /* Block until an event is detected */      
     s_obj.event = get_event();
     ret = smf_run_state(SMF_CTX(&s_obj));
     if (ret) {
          /* handle return code and terminate state machine */
          break;
     }
}
```

This seem's like it would be useful to others and I'll add it to the documentation. Thanks for pointing this out.

closed 2022-08-26 by sambhurst