zephyrproject-rtos/zephyr #48541: subsys/net/l2/ppp/fsm.c: BUS FAULT

PierrickGuitter, 2022-08-01:
**Describe the bug**
Bus Fault during the initialization of gsm modem, using ppp driver.

 - What target platform are you using?
 Custom target with STM32U585 micro and Quectel BG950 gsm modem.
 - What have you tried to diagnose or workaround this issue?
 Debugger and addr2line (see last section for more information)

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b custom_board .  -p
3. west flash
4. See error

**Expected behavior**
No Bus fault should be observed.

**Impact**

**Logs and console output**
```
[00:00:51.155,000] <err> os: ***** BUS FAULT *****
[00:00:51.155,000] <err> os:   Precise data bus error
[00:00:51.155,000] <err> os:   BFAR Address: 0xa4
[00:00:51.155,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000008  r2/a3:  0x00000001
[00:00:51.155,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x08026bfb
[00:00:51.155,000] <err> os:  xpsr:  0x61000000
[00:00:51.155,000] <err> os: r4/v1:  0x00000000  r5/v2:  0x00000000  r6/v3:  0x00000000
[00:00:51.155,000] <err> os: r7/v4:  0x2001ad30  r8/v5:  0x00000000  r9/v6:  0x00000000
[00:00:51.156,000] <err> os: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x2001ad10
[00:00:51.156,000] <err> os: EXC_RETURN: 0xffffffbc
[00:00:51.156,000] <err> os: Faulting instruction address (r15/pc): 0x08025d3c
[00:00:51.156,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:51.156,000] <err> os: Current thread: 0x20003440 (rx_q[0])
[00:00:51.211,000] <err> os: Halting system
```

**Environment :**
 - OS: wsl
 - Zephyr SDK
 - Seen on version zephyr-v3.0.0-1091-gca84f0dcbbed

**Additional context**
Here is the result of the debugging:
addr2line on sp address points to function `struct ppp_context *ppp_fsm_ctx(struct ppp_fsm *fsm)` in fsm.c.
In my context, `fsm` parameter is null, and no protection for null pointer dereference is in place.
The stack trace send back to call of `ppp_send_proto_rej(iface, pkt, protocol);` by `static enum net_verdict process_ppp_msg(struct net_if *iface, struct net_pkt *pkt)` in ppp_l2.c.
`process_ppp_msg`then calls `ppp_send_pkt` with NULL as first parameter `struct ppp_fsm *fsm`.
`ppp_send_pkt` calls incriminated functions `ppp_fsm_ctx` , which dereferences a the NULL pointer fsm.

mmahadevan108, 2022-08-02:
@PierrickGuitter , can you share with us the application you are trying to run so we can debug on different hardware.

PierrickGuitter, 2022-08-04:
@mmahadevan108 The application running is also a custom one.
Essentially, the application :
- Performs the initialization of BG950A-GL (bringing all the pins to their init state)
- Then waits for the modem to be ready and send AT commands for configuration and information collection. It is performed by gsm_configure in the ppp driver.
- Then waits for ppp net interface to be up and ready for sending packets. This is during this step that an IPV6CP-type frame is received and rejected, causing the mentionned bus fault.

I'm aware that this may not be enough for you to reproduce this bug, so please tell me if you want more specific information, or if you want me to share my code.

rlubos, 2022-08-09:
Dereferencing the `fsm` pointer w/o any protection, in case a `NULL` pointer is expected seems clearly wrong, I don't think there's even a need to reproduce. I'll send a fix proposal.

rlubos, 2022-08-09:
@PierrickGuitter Can you give https://github.com/zephyrproject-rtos/zephyr/pull/48830 a shot? Feel free to review as well.

PierrickGuitter, 2022-08-09:
@rlubos Thank you for the fix, it works perfectly on my side.
I see nothing to change in the code for the fix, i'll leave my review with an approvement.

PR #48830 (2022-08-10): net: ppp: Fix NULL pointer dereference in FSM module
ppp_send_pkt() function can be called with NULL fsm parameter (when
PPP_PROTOCOL_REJ packet was sent), howerver this was not taken into
consideration when ppp_context was retrieved. In result, this could lead
to NULL pointer dereference an crash.

Fix this, by moving the ppp_context extraction directly where it's
actually used (PPP_CODE_REJ packet type handling). In such case, fsm
point should not be NULL.

Fixes #48541
closed 2022-08-10 by carlescufi