zephyrproject-rtos/zephyr #48567: MIMXRT1060 custom board support for NXP HAL modules

raghulshanmuganathan-sigmaconnectivity, 2022-08-02:
Hi,

I have created a custom (out-of-tree) board for IMX RT 1061 based on MIMXRT1060_EVK Board. 
I followed all the steps from this guide https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html.

The application builds but does not link to the NXP HAL from modules directory and returns an error: "...  undefined reference to `dcd_data'" whenever I try to configure the CODE_LOCATION.

```
choice CODE_LOCATION
    default CODE_FLEXSPI
endchoice

```

I have also noticed that there is a CMakeLists and hal_nxp.cmake files in modules/hal/nxp/mcux folder which adds more configurations and includes directories based on the known NXP boards.

for example,
```
#include device specific drivers
if (${MCUX_DEVICE} MATCHES "MIMXRT1[0-9][0-9][0-9]")
  zephyr_include_directories(mcux-sdk/devices/${MCUX_DEVICE}/xip)
endif()
```

```
elseif (${MCUX_DEVICE} MATCHES "MIMXRT10[0-9][0-9]")
   include_driver_ifdef(CONFIG_PM_MCUX_GPC        gpc_1        driver_gpc_1)
   include_driver_ifdef(CONFIG_PM_MCUX_DCDC        dcdc_1        driver_dcdc_1)
   include_driver_ifdef(CONFIG_PM_MCUX_PMU        pmu        driver_pmu)
endif()

```

What is the easiest way to add a new custom (out-of-tree) board to these make files? 

I know I could probably create copies of the same files and use them in my board configuration. But it seems like a lot of work to make it work. I was wondering if NXP has a better, clean solution to do this.

ofirshe, 2022-08-03:
following

DerekSnell, 2022-08-05:
Hi @Aragorn.Etiorir ,
As you found, Zephyr support for the NXP boards uses hal_nxp.cmake to include board-specific files into the builds for the in-tree boards.  This includes the dcd.c source file that is missing for your board build.  The board-specific files are pulled in from the MCUXpresso SDK folders in hal_nxp, using the variable MCUX_BOARD.  Once you create your own board and change the board name, this hal_nxp.cmake will no longer find these board-specific files.

These files are board-specific for a reason.  I would want to own those files, and keep them in my custom board folder, in my private repo.  Once located in your board folder, you can modify the CMakeLists.txt file in your board folder to include your board-specific files.  Based on the lines in hal_nxp.cmake using MCUX_BOARD related to RT1xxx, it appears these are the lines you want in your local board file.  Remove the board variables here, and just point to your files/folders:
```
if (${MCUX_BOARD} MATCHES "evk[b]?[bm]imxrt1[0-9][0-9][0-9]")
  list(APPEND CMAKE_MODULE_PATH
    ${CMAKE_CURRENT_LIST_DIR}/mcux-sdk/boards/${MCUX_BOARD}/xip
  )
  include_ifdef(CONFIG_BOOT_FLEXSPI_NOR driver_xip_board_${MCUX_BOARD}${MCUX_BOARD_MOD})
  zephyr_library_sources_ifdef(CONFIG_DEVICE_CONFIGURATION_DATA mcux-sdk/boards/${MCUX_BOARD}/dcd.c)
  zephyr_include_directories(${CMAKE_CURRENT_LIST_DIR}/mcux-sdk/boards/${MCUX_BOARD})
endif()
```
A good example of this is [boards\arm\teensy4\CMakeLists.txt](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/teensy4/CMakeLists.txt).  This is a board using the RT1062 that is not from NXP, and this file includes the local board source files.

Best regards