zephyrproject-rtos/zephyr #48584: netifaces Python package dependency

stephanosio, 2022-08-02:
The `requirements-run-test.txt` file specifies the `netifaces` package as a dependency:

https://github.com/zephyrproject-rtos/zephyr/blob/ce18d3c4ff66ae0f81212ef2f2b9143a1dca6a36/scripts/requirements-run-test.txt#L20-L21

[This package is no longer maintained](https://github.com/al45tair/netifaces) and will likely cause various problems in the future.

It already lacks the pre-built binary package (aka. wheel) for the Python 3.10 release and must be built from source during the package installation (on Windows, this requires users to have Visual Studio installed on their system ...), and it is not going to get any better for the future Python releases.

Consider replacing this dependency with another package that is well maintained.

marc-hb, 2022-08-02:
Perfect timing: for a while I've been asking @enjiamai to get rid of the misguided binding to a semi-random interface and I think he just did: https://github.com/zephyrproject-rtos/zephyr/pull/46972#issuecomment-1199760762

(and to use `INADDR_ANY` by default like any other program
https://docs.python.org/3/library/socket.html#socket-families)

enjiamai, 2022-08-02:
Ok, @stephanosio, I think as @marc-hb mentioned, I will remote the netifaces, thanks.

closed 2022-08-17 by mbolivar-nordic

enjiamai, 2022-08-17:
Thanks for the help! @mbolivar-nordic !

PR #46972 (2022-08-17): soc: xtensa: add support for specifying the service port
This PR includes adding parameters for supporting:

- Specifying the service port for west / twister /hardware map
  (This also change the behavior that west flash have to add --pty {host} to show the output immediately.) 
- Specifying the service port for the cavs server
- Add --tool-opt for the west to pass to cavstool 
- Update the zephyr doc for these changes.
- Remove the dependency of netifaces python package.

**Details:**
Add specifying the port for the remote cavstool remote service.
Ex.
  
```
       west flash --remote-host {h [... cut]