zephyrproject-rtos/zephyr #48608: boards: mps2_an385: Unstable system timer

stephanosio, 2022-08-03:
**Describe the bug**

The system timer on the `mps2_an385` platform, when run inside QEMU with the "icount" mode disabled, is unstable and may return an invalid uptime value (e.g. when calling `k_uptime_get_32()`), leading to frequent test failures and system crashes.

While the `mps2_an385` platform normally [runs with the "icount" mode enabled in order to ensure stable timing](https://github.com/zephyrproject-rtos/zephyr/pull/24811), it still runs the networking tests with the icount mode disabled:

https://github.com/zephyrproject-rtos/zephyr/blob/c11b7852d15c2920303cda813ddd1642c433fb18/boards/Kconfig#L58-L61

This leads to the networking tests randomly failing on the `mps2_an385` platform and forces the twister to retry running these tests until they pass.

While the above has always been an issue, we have recently introduced, in the commit 3d1bbe5b29e6424066efc3363550b5c346fbf7bb, a test case that is more sensitive to the timing stability and this led to a significantly increased test failure rate, to the degree that simply increasing the retry count is no longer acceptable (it is already at 4).

The underlying cause of this timing instability seems to have never been investigated, so investigate and fix this issue accordingly.

**To Reproduce**

Run `tests/kernel/timer/timer_monotonic` on `mps2_an385` with `CONFIG_QEMU_ICOUNT=n`:

```
cmake -GNinja -DBOARD=mps2_an385 -DCONFIG_QEMU_ICOUNT=n ..
ninja run
```

**Logs and console output**

The test fails most of the times; though, it may occasionally succeed.

Attempt 1
```
Running TESTSUITE timer_fn
===================================================================
START - test_timer
k_ticks_to_cyc_floor32(1) = 250000
sys_clock_hw_cycles_per_sec() = 25000000
START - test monotonic timer
diff = -16498779 (t_last = 49847299 : t_now = 33348520);i = 427277

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:80: test_timer: (errors is true)
errors = 1

 FAIL - test_timer in 3.649 seconds
===================================================================
TESTSUITE timer_fn failed.
===================================================================
PROJECT EXECUTION FAILED
```

Attempt 2
```
Running TESTSUITE timer_fn
===================================================================
START - test_timer
k_ticks_to_cyc_floor32(1) = 250000
sys_clock_hw_cycles_per_sec() = 25000000
START - test monotonic timer
diff = -16498882 (t_last = 66367902 : t_now = 49869020);i = 656041

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:80: test_timer: (errors is true)
errors = 1

 FAIL - test_timer in 2.954 seconds
===================================================================
TESTSUITE timer_fn failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Expected behavior**

Test always passes.

**Impact**

Networking tests are rendered ineffective on `mps2_an385` because of an unreasonably high failure rate.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.14.2
 - Commit SHA: bc0d0a390613ca16720441201ba6190c08a974c1

**Additional context**

Related discussion on Discord: https://discord.com/channels/720317445772017664/733037890514321419/1004039756587479091

stephanosio, 2022-08-03:
Marking this low priority since it is a platform-specific issue and the networking tests already run on many other platforms.

stephanosio, 2022-08-03:
cc @andyross @dcpleung @peter-mitsis @microbuilder

andyross, 2022-08-03:
Took a super quick look and reproduced pretty easily.  The smoking gun is in the deltas above:

```
diff = -16498882 (t_last = 66367902 : t_now = 49869020);i = 656041
```

This is using the SysTick driver.  ARM Cortex SysTick is a 24 bit counter[1], the delta there is slipping backwards by 280k cycles less than 2^24.  Given the cycle rate of 25 MHz, that corresponds to 11ms, or just about one host scheduling quantum (i.e. the inevitable timer slop we always have to expect when ICOUNT=n).

Having looked at no code, I'm going to bet that I'm going to go in and find that the driver does do wrap detection, but that it's threshold isn't big enough to handle an interrupt that is 11ms late.  We probably want to scale this to tick rate (i.e. handle up to 1.5 ticks of slop, something like that).

[1] It's actually sorta problematic hardware.  It's a 24 bit down counter which doesn't wrap or otherwise count cycles between when the interrupt is latched and when it's serviced.  So there's a bunch of voodoo tuning that's been done in that driver (not even all of it by me!) trying to get latencies to work right.

andyross, 2022-08-03:
Well, it isn't going to be that easy.  The wraparound logic in cortex_m_systick.c is some fairly subtle code reading a register overflow state flag.  It doesn't look wrong to me at a glance.

Interestingly, the qemu_cortex_m3 platform does **not** show this behavior, even when I set the clock rate to 25Mhz.   Those platform should otherwise be mostly identical (mps2_an385 has the MMU enabled, but this isn't a userspace test).  Might be a good clue to be found in diffing those configs.

Alas I don't know that I'll otherwise have much time to work this in the near term.  I don't know who the best owner of systick is these days?

gramsay0, 2023-09-30:
I think this might be due to a difference in how Zephyr and QEMU expect the timer reload to work.
(I haven't looked into which is correct).

In Zephyr cortex_m_systick.c `sys_clock_set_timeout()` it says that writing `VAL` to zero "resets timer to last_load":

   	SysTick->LOAD = last_load - 1;
	SysTick->VAL = 0; /* resets timer to last_load */

but in QEMU armv7m_systick.c the timer is configured with `PTIMER_POLICY_NO_IMMEDIATE_RELOAD`:

    s->ptimer = ptimer_init(systick_timer_tick, s,
                            PTIMER_POLICY_WRAP_AFTER_ONE_PERIOD |
                            PTIMER_POLICY_NO_COUNTER_ROUND_DOWN |
                            PTIMER_POLICY_NO_IMMEDIATE_RELOAD |
                            PTIMER_POLICY_TRIGGER_ONLY_ON_DECREMENT);

Which does not reload for an additional tick:
 
    /* Starting to run with/setting counter to "0" won't re-load counter
     * immediately, but after a one period.  */
    #define PTIMER_POLICY_NO_IMMEDIATE_RELOAD   (1 << 3)

This causes the time to jump forward by ~ the reload period for single tick and then back again.

I rebuilt QEMU with `PTIMER_POLICY_NO_IMMEDIATE_RELOAD` commented out and the test seemed to pass reliably

gramsay0, 2023-09-30:
I have proposed a fix here:
https://github.com/zephyrproject-rtos/zephyr/pull/63344

closed 2023-10-05 by fabiobaltieri