zephyrproject-rtos/zephyr #48609: drivers: gpio: expose gpio_utils.h to external drivers

mkschreder, 2022-08-03:
**Is your enhancement proposal related to a problem? Please describe.**
The problem is with external gpio drivers. These helpers are needed but by convention everything that is not under "include" is "private" headers.

**Describe the solution you'd like**
I think the best solution would be to implement this as a module instead of a header with a bunch of inline functions. Then expose the module interface under include/drivers/gpio/gpio_utils.h

**Describe alternatives you've considered**
Moving the header into the repo with external drivers. Not optimal because it creates duplication. 

**Additional context**
N/A

balajikulkarni, 2022-08-03:
@mkschreder I would like to pick up this issue. One quick query though, could we have the existing gpio_util.h (with inlines) itself exposed under public header interface?

mkschreder, 2022-08-07:
> @mkschreder I would like to pick up this issue. One quick query though, could we have the existing gpio_util.h (with inlines) itself exposed under public header interface?

Lets start with that. It would be a very small change to start with.

PR #51526 (2022-10-27): drivers: gpio: expose gpio_utils.h to external GPIO drivers
Expose "gpio_utils.h" header for external GPIO drivers.

Fixes #48609.
commit 7f1a5cce (2022-10-27): drivers: gpio: expose gpio_utils.h to external GPIO drivers

Expose "gpio_utils.h" header for external GPIO drivers.

Fixes #48609.
closed 2022-10-27 by carlescufi