zephyrproject-rtos/zephyr #48616: RFC: Change to clang-format coding style rules re binary operators

fgrandel, 2022-08-03:
## Introduction

This proposal is due to a review comment which observed that
```
    if (... something ==
        0)

```    
probably causes more cognitive load than
    
```
    if (... something
        == 0)

```    
for most of us.
    
### Problem description

Especially when operands are very short, breaking after operators may be
hard to read.
    
On the other hand, fixing a single occurrence while not fixing
clang-format rules makes it very hard to maintain files formatted in an
automated way. So unless we put this in the rules file, making such
changes throws the baby out with the bathwater as this essentially
discourages people from using automatic code formatting which is much
worse.
    
### Proposed change

This change therefore proposes a change to formatting rules and
exemplifies it on a few files.

### Concerns and Unresolved Questions

This will obviously introduce whitespace-only changes to all files being
formatted based on the new rules which might introduce unwanted noise
and may break functional blame/praise attribution of code.

As all changes to low-level coding style, such changes tend to provoke
non-productive discussions that hardly contribute to long-term maintainability
of the code.

The rule obviously deviates from the base coding style standard chosen which
may also introduce irritation to those being used to that coding style.

## Alternatives

Do nothing and accept the (partial ugliness)

Not an alternative IMHO: maintain manual changes to formatting and disable
automated formatting.

gmarull, 2022-08-03:
is this about adding `BreakBeforeBinaryOperators: ...` ? Could you provide some examples on how it looks like with/without while formatting a few source files?

fgrandel, 2022-08-03:
Wow, cool, you're so fast I couldn't even link the commit before you asked.

> is this about adding `BreakBeforeBinaryOperators: ...` ? 

Yes that's exactly it - example provided in the attached commit.

gmarull, 2022-08-03:
I think that this change should go into its own PR, and provide a few more examples other than networking code (so we see how it looks like in real code).

fgrandel, 2022-08-03:
Yes, I agree. Let's see whether the original author of the comment also agrees, then I'll prepare a separate PR. @jukkar - can you please comment?

fgrandel, 2022-08-03:
@gmarull Can you propose a few files that I should "tentatively" format? Preferably these should be files that already comply to the formatting rules so that we only see the changes due to BreakBeforeBinaryOperators.

gmarull, 2022-08-03:
> @gmarull Can you propose a few files that I should "tentatively" format? Preferably these should be files that already comply to the formatting rules so that we only see the changes due to BreakBeforeBinaryOperators.

not to reformat anything right now, just provide some examples with option on/off. But anyway, option is clear, so I guess it's ok to just provide the network examples.

jukkar, 2022-08-03:
I just wonder why do we even try to split the checks. Most of the code does not be very readable after that.
For example
```
if (memcmp(dst_plain->addr.ext_addr, ctx->ext_addr, IEEE802154_EXT_ADDR_LENGTH) !=
    0) {
```
vs
```
if (memcmp(dst_plain->addr.ext_addr, ctx->ext_addr,
    IEEE802154_EXT_ADDR_LENGTH) != 0) {
```
IMHO, the latter is much more readable from these two.

fgrandel, 2022-08-03:
@jukkar How can I achieve this with clang-format? This is not compliant with automatic formatting AFAICS...

gmarull, 2022-08-03:
Another question is why clang-format was increased to 100 cols... IMO it should be reverted to 80 cols, or, remove the option so that it gives freedom on where to break (assuming we do an extra check to not exceed 100).

fgrandel, 2022-08-04:
As this is essentially a maintainers' decision, I of course don't have any strong opinions re any of the coding style questions raised in this RFC.

Like probably most of us I'm fine with any coding style as long as there is an automated way to check and enforce it consistently across the whole code base. I'll therefore withdraw from the discussion of details.

Feel free to maintain the RFC open if you think it's relevant. Otherwise no objections from my side to close it.

fgrandel, 2022-09-18:
@gmarull, @jukkar : Do you see any chance that the clang-format line length will be reverted to 80 cols (which I'd support) any time soon or any other change come out of this RFC? Otherwise I propose we close it for inactivity.

fgrandel, 2022-10-13:
Closed, due to inactivity.

closed 2022-10-13 by fgrandel