zephyrproject-rtos/zephyr #48620: LC3 External Source Code

MariuszSkamra, 2022-08-03:
## Origin

liblc3 - Low Complexity Communication Codec library
https://github.com/google/liblc3

## Purpose

Audio codec specified by Bluetooth SIG for LE Audio purposes.

## Mode of integration

External component (fork of existing repository).

## Maintainership

Casper-Bonde-Bose
MariuszSkamra

## Pull Request

TBD

## Description

There is https://github.com/zephyrproject-rtos/liblc3codec repository created already, but it contains copy of old version of the code (without history and modified file structures). Because Zephyr has a copy of the code (not explicit fork) users fork and create issues in Zephyr repository instead of the one the code origins.

## Dependencies

Replacement for https://github.com/zephyrproject-rtos/liblc3codec (should be deprecated ?)

## Revision

e07f313d57347b01c04092dda00515aa6f36c9c8

## License

``Apache-2.0``

stephanosio, 2022-08-03:
cc @nashif @carlescufi

carlescufi, 2022-08-03:
@MariuszSkamra can you provide a sample PR that uses the new repo? For now just point directly to upstream [google/liblc3](https://github.com/google/liblc3).

MariuszSkamra, 2022-08-05:
> @MariuszSkamra can you provide a sample PR that uses the new repo? For now just point directly to upstream [google/liblc3](https://github.com/google/liblc3). This could be the PR that just moves everything from `modules/liblc3codec/` to `modules/liblc3` in the main tree.

Done, just re-opened https://github.com/zephyrproject-rtos/zephyr/pull/46970

carlescufi, 2022-08-10:
TSC approved this today, will create the repo.

carlescufi, 2022-08-12:
Here is the repo: https://github.com/zephyrproject-rtos/liblc3

closed 2022-08-12 by carlescufi