zephyrproject-rtos/zephyr #48625: GSM_PPP api keeps sending commands to muxed AT channel

ivonalo, 2022-08-03:
When enabling the `CONFIG_MODEM_GSM_PPP=y `and ` CONFIG_GSM_MUX=y` options the ` gsm_ppp.c` api keeps sending "AT+CSQ" command to the modem with the `CONFIG_MODEM_GSM_RSSI_POLLING_PERIOD` period. 
Because of this, when I am sending some other AT commands in my custom api to the muxed AT channel, the response to my AT command interferes with the response to the ` AT+CSQ` command and part of my command output gets lost.
 Any idea how to solve this issue of responses interfering?

ivonalo, 2022-08-08:
My question is how to synchronize the access to the modem with the gsm_ppp.c file from the application code? The semaphore is initialized in the gsm_ppp.c file and I have no idea how to access it from my application code

rerickson1, 2022-08-11:
This should not be synchronized from the application code. The driver needs to change to have proper locking to account for this use case.

closed 2022-10-30 by github-actions[bot]