zephyrproject-rtos/zephyr #48665: tests/usb/device: Add zassert to match zassume usage.

aaronemassey, 2022-08-03:
We should nly use `zassume` to check behavior that is already tested elsewhere because `zassume` as it skips the test. We only want to be skipping on conditions that we `zassert` elsewhere. That way we just have a single assertion failure to root cause the test not passing.

The zassume added in https://github.com/zephyrproject-rtos/zephyr/pull/47998/files#diff-3ee7aca813145f447caff74dc876d0a27c14a6a141a3b217a4cc440da6a6b858R198 should have its own small test suite.

Something like:

```
ZTEST(device_usb_enable, test_usb_enable)
{
	zassert_ok(usb_enable(NULL), NULL);
}

/* Small unit test to catch assertion failures on usb_enable failing */
ZTEST_SUITE(device_usb_enable, NULL, NULL, NULL, NULL, NULL);
```
I spent a minute just adding that but got a bind failed error so I figured I should assign this to the API migration author instead as I don't have as much background on the system under test.

@shuai1x Could you please fix this? I can't seem to properly assign the issue to you.

shuai1x, 2022-08-04:
Ok, I know. I will fix it.

aaronemassey, 2022-08-04:
> Ok, I know. I will fix it.

Thank you!

shuai1x, 2022-09-01:
I have opened a PR to fix it. #https://github.com/zephyrproject-rtos/zephyr/pull/49799

closed 2022-09-02 by fabiobaltieri