zephyrproject-rtos/zephyr #48666: driver: ads1x1x: add support for ALERT pin

sslupsky, 2022-08-03:
**Is your enhancement proposal related to a problem? Please describe.**
The device bindings and driver do not have allow configuration of the ALERT pin.  For instance, when the pin is connected to the mcu and signals an interrupt.

**Describe the solution you'd like**
Add something like the following to the binding (the lmp90xxx binding does something like this) and add support in the driver:
```
properties:
    alert-gpios:
      type: phandle-array
      required: false
      description: |
        ALERT pin.

        This pin signals active high when produced by the sensor.  The
        property value should ensure the flags properly describe the
        signal that is presented to the driver.
```

@XenuIsWatching I understand you are the code owner for this driver?  I have some code for handling the alert pin in a separate repo if you want to have a look at that.  I recall I used some code from the lmp90xxx driver.
https://github.com/sslupsky/zephyr-out-of-tree/tree/master/drivers/adc

sslupsky, 2022-08-12:
I found some other issues with this driver.

1.  The binding conflicts with another in tree driver:  ADS1119
2. The driver does not support multiple channels

Also, @galak, the new DT_HAS property in the KConfig file causes conflicts when using an out-of-tree driver.  That is, CMake will compile both the in-tree and out-of-tree drivers and then the linker will fail because of multiple references to the same __device_dts_ord_XXX variable.

XenuIsWatching, 2022-09-05:
> @XenuIsWatching I understand you are the code owner for this driver? I have some code for handling the alert pin in a separate repo if you want to have a look at that. I recall I used some code from the lmp90xxx driver. https://github.com/sslupsky/zephyr-out-of-tree/tree/master/drivers/adc

The reason why I didn't implement the ALERT pin was in the case where I was using it, I was IO limited and didn't have the alert pin connected to the microprocessor I was using.... so it was just left floating
Pull requests are welcome tho...

> 1. The binding conflicts with another in tree driver:  ADS1119

Please give me a link to the link where it is conflicting because I don't see it

> 2. The driver does not support multiple channels

Pull requests are welcome

> @XenuIsWatching I understand you are the code owner for this driver? I have some code for handling the alert pin in a separate repo if you want to have a look at that. I recall I used some code from the lmp90xxx driver.
> https://github.com/sslupsky/zephyr-out-of-tree/tree/master/drivers/adc

Yes, I did take a look at and while reviewing your code and rereading the datasheet. I noticed an issue with the ref_interal and GAIN that would cause an issue with the ads1x13. The ads1113 does have a PGA so it's ref_internal can only be 2048...
I'll make a pull request fixing that sometime this week...

XenuIsWatching, 2022-09-05:
> I'll make a pull request fixing that sometime this week...

#49919