zephyrproject-rtos/zephyr #48682: ADC Support for STM32U575

vibhor-meshram, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
I need to use ADC for my project. I am using nucleo_u575zi_q board. I looked into PR #39202 where ADC support is provided for STM32U585 board. I made necessary changes to my dts files for ADC to work. 
The ADC sample code is now compiling and running as well. I have configured pc0 as adc pin. But the output I am getting is floating values irrespective of what I connect at ADC input(VCC/GND). 

**Results**
`ADC reading:
- ADC_1, channel 1: 296 = 59 mV
ADC reading:
- ADC_1, channel 1: 292 = 58 mV
ADC reading:
- ADC_1, channel 1: 295 = 59 mV
ADC reading:
- ADC_1, channel 1: 293 = 59 mV
ADC reading:
- ADC_1, channel 1: 293 = 59 mV
ADC reading:
- ADC_1, channel 1: 302 = 60 mV
ADC reading:
- ADC_1, channel 1: 301 = 60 mV
ADC reading:
- ADC_1, channel 1: 244 = 49 mV
ADC reading:
- ADC_1, channel 1: 300 = 60 mV
ADC reading:
- ADC_1, channel 1: 296 = 59 mV
ADC reading:
- ADC_1, channel 1: 296 = 59 mV
ADC reading:
- ADC_1, channel 1: 298 = 60 mV`

**Describe the solution you'd like**
The output of ADC must change as per the input. 

**Describe alternatives you've considered**
Added adc node to dts file by referring to STM32U585 board. Is there anything else needed to be done?

**Additional context**
Attached: DTS file, overlay file
[nucleo_u575zi_q_DTS_File.txt](https://github.com/zephyrproject-rtos/zephyr/files/9258860/nucleo_u575zi_q_DTS_File.txt)
[nucleo_u575zi_q_overlay_file.txt](https://github.com/zephyrproject-rtos/zephyr/files/9258862/nucleo_u575zi_q_overlay_file.txt)
.

erwango, 2022-08-05:
I had a test on b_u585i_iot02a and I'm seeing same behavior. Converting to bug

FRASTM, 2022-08-08:
@vibhor-meshram could you please check with the https://github.com/zephyrproject-rtos/zephyr/pull/48770

vibhor-meshram, 2022-08-09:
ADC works using above fix.

closed 2022-08-09 by vibhor-meshram