zephyrproject-rtos/zephyr #48722: flash_map: pointer dereferencing causes build to fail

sslupsky, 2022-08-05:
**Describe the bug**
A recent change to flash_map is causing build problems.  A pointer is not properly declared which causes the build to fail.
https://github.com/zephyrproject-rtos/zephyr/blob/57487622f57dbb3aa7582f8dbf62c148e07c9138/subsys/storage/flash_map/flash_map_shell.c#L26-L28

```
In file included from /Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/storage/flash_map/flash_map_shell.c:7:
/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/storage/flash_map/flash_map_shell.c: In function 'fa_cb':
/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/storage/flash_map/flash_map_shell.c:27:46: error: dereferencing pointer to incomplete type 'const struct device'
       fa->fa_id, fa->fa_device_id, fa->fa_dev->name,
                                              ^~
```

flash_map_shell.c and flash_map.h do not include device.h so the device struct has an incomplete type hence the error.  Including device.h in flash_map.h resolves the problem.

@JordanYates  FYI.

sslupsky, 2022-08-05:
Also, the fa->fa_off argument is throwing a Wformat warning.  I am compiling on arm32 and the sizes are actually correct but the compiler is generating a warning.  Probably should type cast the argument?

fabiobaltieri, 2022-08-09:
FYI @de-nordic.

PR #49191 (2022-08-18): storage: flash_map: fix incomplete type
A pointer is not properly declared properly which causes the
build to fail.

flash_map_shell.c and flash_map.h do not include device.h so
the device struct has an incomplete type hence the build error.
Including device.h resolves the problem.

Also fixes a Wformat warning when referencing fa_off which is a pointer.
Cast the pointer to an uint32_t.

Fixes #48722
commit fd142362 (2022-08-18): storage: flash_map: fix incomplete type

A pointer is not properly declared properly which causes the
build to fail.

flash_map_shell.c and flash_map.h do not include device.h so
the device struct has an incomplete type hence the build error.
Including device.h resolves the problem.

Also fixes a Wformat warning when referencing fa_off which is a pointer.
Cast the pointer to an uint32_t.

Fixes #48722
closed 2022-08-18 by carlescufi