zephyrproject-rtos/zephyr #48724: mpu9250 driver init function register setup using the same config parameter twice.

ShenMansell, 2022-08-05:
the mpu9250.c driver mpu9250_init function seems suspect at line 311 & 312 ->
```
ret = i2c_reg_write_byte_dt(&cfg->i2c, MPU9250_REG_ACCEL_CFG2,
                    cfg->gyro_dlpf);
```
It is using the gyro_dlpf to setup the accel cfg register. Presumably this should be using cfg->accel_dlpf?

This seems like a typo, and that thinking is confirmed by
* checks to the cfg->accel_dlpf directly before this line.
* cfg->gyro_dlpf is used to setup a different register earlier in the function.
* cfg->accel_dlpf is not used to setup any registers.

gmarull, 2022-08-05:
cc: @dchat-nordic

fabiobaltieri, 2022-08-09:
@ShenMansell can you send a pull request?

PR #49217 (2022-08-18): drivers: sensor: mpu9250: Fix initialization routine
Fix a call that writes Register 29 - Accelerometer Configuration 2.
Due to an obvious copy-paste mistake in the code, this register was
written with a DLPF setting from devicetree intended for gyroscope,
not for accelerometer (and the latter was effectively ignored).

Fixes #48724.
closed 2022-08-18 by carlescufi