zephyrproject-rtos/zephyr #48725: arm_thread_swap:  tests/arch/arm/arm_thread_swap/ failed on reel_board

smrtos, 2022-08-05:
**Describe the bug**

Run below command:
`twister -vv --hardware-map ~/daily_qa/cavs25.map --device-testing -s tests/arch/arm/arm_thread_swap/arch.arm.swap.common.fpu_sharing.no_optimizations`

Saw failure:
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-3122-g3ba0cb19ea99  ***
DEBUG   - DEVICE: Running TESTSUITE arm_thread_swap
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: S: ***** MPU FAULT *****
DEBUG   - DEVICE: E:   Stacking error (context area might be not valid)
DEBUG   - DEVICE: E:   Data Access Violation
DEBUG   - DEVICE: E:   MMFAR Address: 0x20003034
DEBUG   - DEVICE: E: r0/a1:  0x0001c8dd  r1/a2:  0x0001c8dd  r2/a3:  0x0001c8dd
DEBUG   - DEVICE: E: r3/a4:  0x0001c8dd r12/ip:  0x0001c8dd r14/lr:  0x0001c8dd
DEBUG   - DEVICE: E:  xpsr:  0x0001c800
DEBUG   - DEVICE: E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
DEBUG   - DEVICE: E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
DEBUG   - DEVICE: E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
DEBUG   - DEVICE: E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
DEBUG   - DEVICE: E: fpscr:  0x00000002
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x0001c8dd
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x20000328 (main)
DEBUG   - DEVICE: E: Halting system
DEBUG   - -- west flash: using runner pyocd
```

Bisect leads to this commit: bfd59b17a1913ee00cbe447b59cdfd514eeeb693

fabiobaltieri, 2022-08-09:
@shaoanxli can you look into this? (it got bisected to https://github.com/zephyrproject-rtos/zephyr/pull/47525)

microbuilder, 2022-08-09:
Looks like this test simply needs more stack memory than the default 1K due to no optimisations. Rebase should fix this with https://github.com/zephyrproject-rtos/zephyr/pull/48776

microbuilder, 2022-08-09:
@fabiobaltieri Do you have a reel board to test this with? As per comment above, it should now be fixed.

microbuilder, 2022-08-09:
Resolved via https://github.com/zephyrproject-rtos/zephyr/pull/48776

closed 2022-08-09 by microbuilder