zephyrproject-rtos/zephyr #48726: net: tests/net/ieee802154/l2/net.ieee802154.l2 failed on reel board

smrtos, 2022-08-05:
**Describe the bug**
Run:
`twister -vv --hardware-map ~/daily_qa/cavs25.map --device-testing -s tests/net/ieee802154/6lo_fragment/net.ieee802154.fragment`

Failure:
```
DEBUG   - DEVICE: E: ***** BUS FAULT *****
DEBUG   - DEVICE: E:   Precise data bus error
DEBUG   - DEVICE: E:   BFAR Address: 0x4f48088c
DEBUG   - DEVICE: E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0xffffffff
DEBUG   - DEVICE: E: r3/a4:  0x00002771 r12/ip:  0x0000001f r14/lr:  0x0000aa57
DEBUG   - DEVICE: E:  xpsr:  0x61000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x0000aa60
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x200019b8 (main)
DEBUG   - DEVICE: E: Halting system
```

Hardware map like this:
```
- connected: true
  id: xxxxxxx
  platform: reel_board
  product: DAPLink CMSIS-DAP
  runner: pyocd
  serial: /dev/ttyACM0
```

Bisect leads to: f0441fba58a242d4fcc70794f6250ace358baae4

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

galak, 2022-08-05:
The only difference that SHA would make is enabling a few extra sensor drivers.

Can you try disabling the `ti_hdc` sensor either via Kconfig or devicetree.

galak, 2022-08-05:
@rlubos does it make sense that `NRF_802154_TEMPERATURE_UPDATE` is `default y`??

rlubos, 2022-08-05:
> @rlubos does it make sense that `NRF_802154_TEMPERATURE_UPDATE` is `default y`??

@jciupis How does it look from the driver perspective? I wonder if the config should be enabled implicitly given it has dependencies to a temperature sensor.

jciupis, 2022-08-08:
@rlubos I think it would be good to keep `NRF_802154_TEMPERATURE_UPDATE` enabled by default. Temperature affects RSSI, so disabling it will lead to inaccurate RSSI readouts.

galak, 2022-08-24:
@smrtos any ability to try what I suggested?

smrtos, 2022-08-25:
@galak  Will try it. Sorry for the late response.

shuai1x, 2022-08-26:
> Can you try disabling the `ti_hdc` sensor either via Kconfig or devicetree.
We have tried to disable the ti_hdc` sensor in every prj.conf under the tests/net/ieee802154. But it would see the same failure like before. And the CONFIG_TI_HDC didn't exist in .config after I modify the prj.conf.

smrtos, 2022-09-23:
@shuai1x  Could you be more specific about what you have tried?

shuai1x, 2022-10-14:
I have tested it again. And see the build failure. Link to the issue [https://github.com/zephyrproject-rtos/zephyr/issues/50732](url).

galak, 2022-10-18:
Can you try setting CONFIG_NET_IF_MAX_IPV6_COUNT=2

shuai1x, 2022-10-19:
> Can you try setting CONFIG_NET_IF_MAX_IPV6_COUNT=2

**Before:**
Steps:
```
vim tests/net/ieee802154/l2/testcase.yaml (delete platform_allow)
west build -p auto -b reel_board tests/net/ieee802154/l2
see error
```
Error:
```
-- west build: building application
[1/377] Preparing syscall dependency handling

[2/377] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/zhao/zephyrproject/zephyr), build: zephyr-v3.2.0-512-gb09973c460e3
[335/377] Building C object modules/mbedtls/CMakeFiles/module...e/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c.obj
/home/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c: In function 'myrand':
/home/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c:2356:21: warning: implicit declaration of function 'rand'; did you mean 'myrand'? [-Wimplicit-function-declaration]
 2356 |         output[i] = rand();
      |                     ^~~~
      |                     myrand
[367/377] Linking C executable zephyr/zephyr_pre0.elf

[371/377] Linking C executable zephyr/zephyr_pre1.elf

[377/377] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      179396 B         1 MB     17.11%
             RAM:       57548 B       256 KB     21.95%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**After setting CONFIG_NET_IF_MAX_IPV6_COUNT=2:**
Steps:
```
vim tests/net/ieee802154/l2/testcase.yaml (delete platform_allow, add CONFIG_NET_IF_MAX_IPV6_COUNT=2)
west build -p auto -b reel_board tests/net/ieee802154/l2
see error
```
Error:
```
-- west build: building application
[1/377] Preparing syscall dependency handling

[2/377] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/zhao/zephyrproject/zephyr), build: zephyr-v3.2.0-708-g3cef8e68df3e
[311/377] Building C object modules/mbedtls/CMakeFiles/module...e/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c.obj
/home/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c: In function 'myrand':
/home/zhao/zephyrproject/modules/crypto/mbedtls/library/rsa.c:2356:21: warning: implicit declaration of function 'rand'; did you mean 'myrand'? [-Wimplicit-function-declaration]
 2356 |         output[i] = rand();
      |                     ^~~~
      |                     myrand
[367/377] Linking C executable zephyr/zephyr_pre0.elf

[371/377] Linking C executable zephyr/zephyr_pre1.elf

[377/377] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      180388 B         1 MB     17.20%
             RAM:       57724 B       256 KB     22.02%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**Summary:**
It seems the same result after setting CONFIG_NET_IF_MAX_IPV6_COUNT=2. And this issue can't be reproduced. Now tests/net/ieee802154/l2 failed due to build failure. You can see the detail from the new issue #https://github.com/zephyrproject-rtos/zephyr/issues/50732

shuai1x, 2022-10-21:
> @shuai1x can you this again with CONFIG_NET_IF_MAX_IPV6_COUNT=2. The build issue with rand() should now be fixed.

I have tested it again. The build failure has been solved by PR #https://github.com/zephyrproject-rtos/zephyr/pull/51479. After that it can pass on reel board and can't reproduce this issue. So it can't influence this test whether add CONFIG_NET_IF_MAX_IPV6_COUNT=2 or not. So I think this issue can be closed.

closed 2022-10-21 by galak