zephyrproject-rtos/zephyr #48731: gen_handles script fails with pwmleds handle

nordicjm, 2022-08-05:
**Describe the bug**
The updates to the gen_handles scripts which adds support for graphviz output seems to fail in specific circumstances, a project for example has this contents in the dev_handles.c file:
```
/* 1 : /soc/clock@40000000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_clock_40000000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 2 : /soc/gpio@50000300:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_gpio_50000300[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 3 : /soc/gpio@50000000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_gpio_50000000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 4 : /soc/uart@40028000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_uart_40028000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 5 : /soc/uart@40002000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_uart_40002000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 6 : /soc/pwm@4002d000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_pwm_4002d000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 7 : /soc/pwm@40022000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_pwm_40022000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 8 : /soc/pwm@40021000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_pwm_40021000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 9 : /soc/pwm@4001c000:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_soc_S_pwm_4001c000[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };

/* 10 : /pwmleds:
 */
const device_handle_t __aligned(2) __attribute__((__section__(".__device_handles_pass2")))
__devicehdl_DT_N_S_pwmleds[] = { DEVICE_HANDLE_SEP, DEVICE_HANDLE_SEP, DEVICE_HANDLE_ENDS };
```

The script fails on the last part due to label on line 271 of scripts/build/elf_parser.py being none:
```
cd nrf/samples/caf/_AA/zephyr && /usr/bin/python3.10 zephyr/scripts/build/gen_handles.py --output-source dev_handles.c --output-graphviz dev_graph.dot --num-dynamic-devices 0 --kernel nrf/samples/caf/_AA/zephyr/zephyr_pre0.elf --zephyr-base zephyr --start-symbol __device_start
Traceback (most recent call last):
  File "zephyr/scripts/build/gen_handles.py", line 141, in <module>
    main()
  File "zephyr/scripts/build/gen_handles.py", line 124, in main
    dot = parsed_elf.device_dependency_graph('Device dependency graph', args.kernel)
  File "zephyr/scripts/build/elf_parser.py", line 272, in device_dependency_graph
    text = '{:s}\\nOrdinal: {:d} | Handle: {:d}\\n{:s}'.format(
TypeError: unsupported format string passed to NoneType.__format__
ninja: build stopped: subcommand failed.
```

A workaround is to check if the label type is none, but I don't think that is the correct solution. Reverting commit 29942475c5ce4c9f1d7a76273912777445f7d429 allows the script to work

**To Reproduce**
Requires use of the nRF connect SDK to get the features required to exhibit this issue, but getting https://github.com/nrfconnect/sdk-nrf/pull/8327 with west and then trying to build `nrf/samples/caf` for the `nrf52840dk_nrf52840` board shows the issue

**Expected behavior**
No error or script failure

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.0-rc1
 - Commit SHA or Version used: For zephyr, using commit 71ef669ea4a73495b255f27024bcd5d542bf038c

gmarull, 2022-08-05:
This has been fixed https://github.com/zephyrproject-rtos/zephyr/commit/700745195525f311fd939fad4fe8cd2bdda17dc3

nordicjm, 2022-08-05:
> This has been fixed [7007451](https://github.com/zephyrproject-rtos/zephyr/commit/700745195525f311fd939fad4fe8cd2bdda17dc3)

Wonderful! Thanks

closed 2022-08-05 by nordicjm