zephyrproject-rtos/zephyr #48738: dts: label: label defined in soc does not take effects in final zephyr.dts

hakehuang, 2022-08-05:
**Describe the bug**

label defined in dts\arm\nxp\nxp_rt11xx.dtsi, does not take effects in final zephyr.dts

Please also mention any information which could help others to understand

we have defined i2c1 label in zephyr\dts\arm\nxp\nxp_rt11xx.dtsi

```
		lpi2c1: i2c@40104000 {
			compatible = "nxp,imx-lpi2c";
			clock-frequency = <I2C_BITRATE_STANDARD>;
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40104000 0x4000>;
			interrupts = <32 0>;
			clocks = <&ccm IMX_CCM_LPI2C1_CLK 0x70 6>;
			label = "I2C_1";
			status = "disabled";
		};
```

and after we build a application the label is not passed to final zephyr.dts

```
                lpi2c1: i2c@40104000 {
                        compatible = "nxp,imx-lpi2c";
                        clock-frequency = < 0x186a0 >;
                        #address-cells = < 0x1 >;
                        #size-cells = < 0x0 >;
                        reg = < 0x40104000 0x4000 >;
                        interrupts = < 0x20 0x0 >;
                        clocks = < &ccm 0x400 0x70 0x6 >;
                        status = "okay";
                        pinctrl-0 = < &pinmux_lpi2c5 >;
                        pinctrl-names = "default";
                };
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/sensor/fxos8700
2. west build -b mimxrt1170_evk_cm7

**Expected behavior**
label is passed to final dts so application can use the label to get driver device_get_binding(label)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 1e25763f572309c7e92bcbb3461d8263b1006245

galak, 2022-08-06:
Tried reproducing this and I see the `label` property in <BUILD>/zephyr/zephyr.dts

hakehuang, 2022-08-06:
@galak  can you attach your zephyr.dts? my result seems different
[zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/9275287/zephyr.zip)

galak, 2022-08-06:
Here's me zephyr.dts [zephyr.dts.gz](https://github.com/zephyrproject-rtos/zephyr/files/9275337/zephyr.dts.gz)

Here's my tweak to dts:
```
diff --git a/dts/arm/nxp/nxp_rt11xx.dtsi b/dts/arm/nxp/nxp_rt11xx.dtsi
index 854da3a461..4a2c50ba89 100644
--- a/dts/arm/nxp/nxp_rt11xx.dtsi
+++ b/dts/arm/nxp/nxp_rt11xx.dtsi
@@ -262,6 +262,7 @@
                        interrupts = <32 0>;
                        clocks = <&ccm IMX_CCM_LPI2C1_CLK 0x70 6>;
                        status = "disabled";
+                       label = "KUMAR";
                };
 
                lpi2c2: i2c@40108000 {
```

hakehuang, 2022-08-06:
@galak , the dts/arm/nxp/nxp_rt11xx.dtsi content is as below, there is already label defined.

```
		lpi2c1: i2c@40104000 {
			compatible = "nxp,imx-lpi2c";
			clock-frequency = <I2C_BITRATE_STANDARD>;
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40104000 0x4000>;
			interrupts = <32 0>;
			clocks = <&ccm IMX_CCM_LPI2C1_CLK 0x70 6>;
			label = "I2C_1";
			status = "disabled";
		};
```

galak, 2022-08-06:
> @galak , the dts/arm/nxp/nxp_rt11xx.dtsi content is as below, there is already label defined. I guess we are not in the same code base.
> 
> ```
> 		lpi2c1: i2c@40104000 {
> 			compatible = "nxp,imx-lpi2c";
> 			clock-frequency = <I2C_BITRATE_STANDARD>;
> 			#address-cells = <1>;
> 			#size-cells = <0>;
> 			reg = <0x40104000 0x4000>;
> 			interrupts = <32 0>;
> 			clocks = <&ccm IMX_CCM_LPI2C1_CLK 0x70 6>;
> 			label = "I2C_1";
> 			status = "disabled";
> 		};
> ```

We recently removed that in upstream Zephyr.

hakehuang, 2022-08-07:
@ok, seems it is fixed in latest version

closed 2022-08-07 by hakehuang