zephyrproject-rtos/zephyr #48742: Linking fails during build when referencing functions in `zephyr/bluetooth/crypto.h`

atoncetti, 2022-08-05:
**Describe the bug**
Linking fails during build when referencing functions in `zephyr/bluetooth/crypto.h` with the following error:

```
[224/234] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/antonio/Software/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.o -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.o  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/antonio/Software/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/nordic_nrf/nrf51/libsoc__arm__nordic_nrf__nrf51.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/sensor/fxos8700/libdrivers__sensor__fxos8700.a  zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v6-m/nofp"  -L/home/antonio/Software/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,--orphan-handling=warn && cd /home/antonio/Software/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
/opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(main.c.o): in function `main':
/home/antonio/Projects/claimore/bt_cypto_example/src/main.c:23: undefined reference to `bt_ccm_decrypt'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/antonio/Software/zephyrproject/zephyr/build
```

Here is the relevant part of the code:
```
#include <zephyr/zephyr.h>
#include <zephyr/bluetooth/crypto.h>

void main(void)
{
    int rc = 0;
    uint8_t subkey[16];
    char plaintext[16];
    uint8_t salt[16];
    char request[16];
    
    rc = bt_ccm_decrypt(subkey, salt, request, 16, NULL, 0, plaintext, 0);
    
    return;
}
```

and `prj.conf`:
```
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_LOG_MODE_MINIMAL=y
```

I am trying to build for a `bbc_microbit` with `west build -p auto -b bbc_microbit ./bt_cypto_example`

Am I missing a config flag?

A minimal example can be found at: https://github.com/atoncetti/bt_crypto_example

Thank you for your input.

**Environment:**
- Zephyr v3.1.0
- Zephyr SDK v0.14.2
- Ubuntu 22.04

hermabe, 2022-08-08:
It looks like the functions in `aes_ccm.c` are only linked in if `CONFIG_BT_HOST_CCM=y`.

atoncetti, 2022-08-08:
Thank you @hermabe, that was indeed the issue! 

Now it builds correctly when I add the flag `CONFIG_BT_HOST_CCM=y`.

Closing this for the solution above.