zephyrproject-rtos/zephyr #48757: Windows10 Installation: Failed to run `west update`

r0n9Y, 2022-08-06:
**Failed to run `west update` when I try to install Zephyr SDK first time**

```
=== updating zscilib (modules/lib/zscilib):
west.manifest: zscilib: checking if cloned
west.manifest: running 'git rev-parse --show-cdup' in C:\Users\sanju\zephyr_project\modules\lib\zscilib
west.manifest: "git rev-parse --show-cdup" exit code: 0 stdout: b'\n' stderr: b''
west.manifest: running 'git cat-file -t a54986aa98db4082ac56b582843bb5b5435208a6' in C:\Users\sanju\zephyr_project\modules\lib\zscilib
west.manifest: "git cat-file -t a54986aa98db4082ac56b582843bb5b5435208a6" exit code: 0 stdout: b'commit\n' stderr: b''
west.manifest: running 'git rev-parse --verify --symbolic-full-name a54986aa98db4082ac56b582843bb5b5435208a6' in C:\Users\sanju\zephyr_project\modules\lib\zscilib
west.manifest: "git rev-parse --verify --symbolic-full-name a54986aa98db4082ac56b582843bb5b5435208a6" exit code: 0 stdout: b'' stderr: b''
skipping unnecessary fetch
west.manifest: running 'git update-ref -m 'west update: moving to a54986aa98db4082ac56b582843bb5b5435208a6^{commit}' refs/heads/manifest-rev 'a54986aa98db4082ac56b582843bb5b5435208a6^{commit}'' in C:\Users\sanju\zephyr_project\modules\lib\zscilib
fatal: a54986aa98db4082ac56b582843bb5b5435208a6^commit: not a valid SHA1
west.manifest: "git update-ref -m 'west update: moving to a54986aa98db4082ac56b582843bb5b5435208a6^{commit}' refs/heads/manifest-rev 'a54986aa98db4082ac56b582843bb5b5435208a6^{commit}'" exit code: 128 stdout: None stderr: None
ERROR: update failed for multiple projects; see above
```
 - Zephyr version: 3.1.99
 - Following the installation guide for window from https://docs.zephyrproject.org/latest/develop/getting_started/index.html
 - Unable to get the Zephyr source code
 - env: Windows 10 version: 21H2 19044.1826

carlescufi, 2022-08-08:
@r0n9Y which west version are you using? `west --version`

carlescufi, 2022-08-08:
> Is it possible I can find pre-built binary somewhere? It is unusual that the ppl have to build the SDK from scratch. And I am trying to avoid to use PlatformIO with Zephyr.

You do not need to build the SDK from scratch, this is Zephyr and its source code and not the SDK.

r0n9Y, 2022-08-08:
West version: v0.13.1