zephyrproject-rtos/zephyr #48779: net.socket.select: failed (qemu/mps2_an385)

erwango, 2022-08-08:
**Describe the bug**
See https://github.com/zephyrproject-rtos/zephyr/runs/7721882961?check_suite_focus=true

`tests/net/socket/tcp/net.socket.tcp (mps2_an385:tests/net/socket/tcp/net.socket.tcp) failed`.

Error log: 
```
START - test_v4_msg_waitall

    Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tcp/src/main.c:203: test_context_cleanup: (wait_for_n_tcp_contexts(0, TCP_TEARDOWN_TIMEOUT) not equal to 0)
Not all TCP contexts properly cleaned up
 FAIL - test_v4_msg_waitall in 0.200 seconds
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Block CI

stephanosio, 2022-08-08:
Likely related to https://github.com/zephyrproject-rtos/zephyr/issues/48608

rlubos, 2022-08-08:
I'd say we disable other time-sensitive network tests on this platform as well, until https://github.com/zephyrproject-rtos/zephyr/issues/48608 is resolved. I'll send a PR.

PR #48781 (2022-08-08): tests: net: socket: Disable unstable tests for mps2_an385
Disable unstable tests cases for mps2_an385, which fail sporadically
due to timer stability issues on that platform (see #48608).

Fixes #48779
closed 2022-08-08 by stephanosio