zephyrproject-rtos/zephyr #48813: bt_conn_disconnect randomly gives error "bt_conn: not connected!"

Klievan, 2022-08-09:
**Describe the bug**
When calling bt_conn_disconnect() in a BLE-characteristic write-callback I get randomly returned the following error:

```
[00:00:22.404,479] <inf> main: Data cleared. Disconnecting...
[00:00:22.454,803] <err> bt_conn: not connected!
[00:00:22.455,139] <wrn> bt_hci_core: opcode 0x2013 status 0x02
[00:00:22.455,535] <inf> main: Disconnected: C5:DC:C7:91:04:9B (random) (reason 0x16)
```

Stripped down function where I call bt_conn_disconnect():
```
static void on_data_reset_cb(struct bt_prx_client *prx_c,
									 int err)
{
	if (err)
	{
		LOG_ERR("Failed to reset data in callback.");
		bt_conn_disconnect(prx_c->conn, BT_HCI_ERR_REMOTE_USER_TERM_CONN);
	}
	else
	{

			LOG_INF("Data cleared. Clock has not been received or is stale. Disconnecting...");
			bt_conn_disconnect(prx_c->conn, BT_HCI_ERR_REMOTE_USER_TERM_CONN);
	}
}
```

When enabling debug logs using the below prj.conf options the error seems to dissapear. This makes me believe that there is some sort of timing-issue.
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_DEBUG_CONN=y

The call-stack of the "bt_conn: not connected!" error-print in OZone is the following: 
[image]

Using OZone I can see that in the code below (responsible for printing the connection-error) the connection state is currently `BT_CONN_DISCONNECTING`
```
if (conn->state != BT_CONN_CONNECTED) {
    BT_ERR("not connected!");
    return -ENOTCONN;
  }
```

At first glance it seems like the application is trying to update the connection parameters when the connection is currently disconnecting and thus giving an error. Not entirely sure why that is happening, I have `CONFIG_BT_GAP_AUTO_UPDATE_CONN_PARAMS=n` set

I noticed the same ble_hci_core warning in issue https://github.com/zephyrproject-rtos/zephyr/issues/48505 . Not sure if they are related?

**Impact**
Currently the error seems to be harmless and can be ignored, although we have not run this application for a considerable amount of time.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: NCS - Zephyr
 - Zephyr OS build v3.0.99-ncs1

Klievan, 2022-08-09:
```
13150	105.396	Master_0x8c4be0c9	LE 1M	LE LL	0	150µs	0	0	False	193	Empty PDU
13151	105.397	Slave_0x8c4be0c9	LE 1M	ATT	7	150µs	0	1	False	193	Rcvd Handle Value Notification, Handle: 0x001a (Unknown: Unknown)
13152	105.419	Master_0x8c4be0c9	LE 1M	ATT	7	22750µs	1	1	False	194	Sent Write Request, Handle: 0x001a (Unknown: Unknown)
13153	105.420	Slave_0x8c4be0c9	LE 1M	LE LL	0	150µs	1	0	False	194	Empty PDU
13154	105.444	Master_0x8c4be0c9	LE 1M	LE LL	0	24634µs	0	0	False	195	Empty PDU
13155	105.445	Slave_0x8c4be0c9	LE 1M	ATT	5	150µs	0	1	False	195	Rcvd Write Response, Handle: 0x001a (Unknown: Unknown)
13156	105.469	Master_0x8c4be0c9	LE 1M	ATT	11	24650µs	1	1	False	196	Sent Write Request, Handle: 0x0017 (Unknown: Unknown)
13157	105.470	Slave_0x8c4be0c9	LE 1M	LE LL	0	150µs	1	0	False	196	Empty PDU
13158	105.494	Master_0x8c4be0c9	LE 1M	LE LL	0	24603µs	0	0	False	197	Empty PDU
13159	105.495	Slave_0x8c4be0c9	LE 1M	ATT	5	150µs	0	1	False	197	Rcvd Write Response, Handle: 0x0017 (Unknown: Unknown)
13160	105.519	Master_0x8c4be0c9	LE 1M	LE LL	0	24650µs	1	1	False	198	Empty PDU
13161	105.520	Slave_0x8c4be0c9	LE 1M	LE LL	0	150µs	1	0	False	198	Empty PDU
13162	105.569	Master_0x8c4be0c9	LE 1M	LE LL	2	49691µs	0	0	False	200	Control Opcode: LL_TERMINATE_IND
13163	105.570	Slave_0x8c4be0c9	LE 1M	L2CAP	16	150µs	0	1	False	200	Connection Parameter Update Request
```

carlescufi, 2022-08-11:
It looks like the peer sends a `BT_L2CAP_CONN_PARAM_REQ` at the same time we disconnect locally. This causes the sending of the l2cap conn param response to fail because the connection is no longer active. This is not really a bug, suggestion is to change the `BT_ERR("not connected!");` to `BT_WRN("not connected!");`, but needs more discussion. @Klievan feel free to send a PR to start a discussion.

commit 71b4dcf2 (2022-08-17): Bluetooth: L2CAP: Add check to see if device is not disconnected

Currently, if a bluetooth peripheral sends a connection parameter update
request to a Zephyr central while the central is in the process of
disconnecting and has sent a connection terminate command in the same
connection interval, the central will try to reply, resulting in an
error being printed in the console.

This commit adds a check in le_conn_param_update_req that catches this
state before the central considers this request, preventing the unneeded
error being printed.

Fixes #48813
closed 2022-08-17 by fabiobaltieri