zephyrproject-rtos/zephyr #48841: Bluetooth: df: Assert in lower link layer when requesting CTE from peer periodically with 7.5ms connection interval

ppryga, 2022-08-09:
**Describe the bug**
There is an `LL_ASSERT` that fires: https://github.com/zephyrproject-rtos/zephyr/blob/a150c7ee77687d543b903ea9008754da7415f4b5/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_df.c#L246-L247

Conditions:
1. Connected peer device that supports CTE RSP.
2. Connection interval is 7.5ms
3. Local device requests periodically CTE. Request interval is 1 connection event, hence CTE_REQ control procedure starts every connection event.
4. The LL_ASSERT fires because there is not enough `node_rx_iq_report` in a pool to handle so frequent IQ report generation.

**To Reproduce**
Steps to reproduce the behavior:
1. Use sample `direction_finding_central` and  change it to use conn interval 7.5ms and CTE_REQ every connection event.
2. Use sample `direction_finding_peripheral` as a peer device.
6. Observe central output how it fails with assert.

**Expected behavior**
No LL_ASSERT fires.
 
**Impact**
Low. Happens only when DF is enabled in very specific configuration. Small increase to connection interval should help handle the issue.  

**Additional context**
The proper way to handle the issue is to send an IQ report with `Pocket_Status=0xFF` and `Sample_Count=0x0`

cvinayak, 2022-09-20:
@ppryga-nordic will #48936 fix this issue?

ppryga-nordic, 2022-09-20:
@cvinayak No, the PR https://github.com/zephyrproject-rtos/zephyr/pull/48936 handles insufficient `node_rx_iq_report` for periodic scanner only.
There part for DF connected mode should be implemented. 

Actually I suggest a change in the PR. Witch changs to `lll_df_conf_cte_rx_enable` that remove `LL_ASSERT` the connected mode will execute reception of CTE even there is no memory for IQ report. Hence proper handling this situation should be implemented in the PR or value returned by `lll_df_conf_cte_rx_enable` should be checked in `lll_conn.c` and `lll_peripheral.c`.

closed 2025-01-26 by github-actions[bot]