zephyrproject-rtos/zephyr #48857: samples: Bluetooth: Buffer size mismatch in samples/bluetooth/hci_usb for nRF5340

Thalley, 2022-08-10:
**Describe the bug**
The `samples/bluetooth/hci_usb` will generally use the default buffer sizes, e.g. 27 for `BT_BUF_ACL_TX_SIZE` and 27 for `BT_BUF_ACL_RX_SIZE`, when compiled for the appcore on the nRF5340.

When this sample is then used with e.g. `native_posix` with `BT_SMP=y`, then the `hci_usb` sample will no longer work, due to these sizes, if the `native_posix` host build will attempt to use e.g. LE Secure connections. This is due to the `hci_usb` sample using default buffer sizes of 27, but the `native_posix` build will attempt to use 69 for the `BT_BUF_ACL_RX_SIZE` to accommodate the public key exchange in LE Secure Connections, if the controller reports buffer sizes of above or equal to 69 in the Read Buffer Size command. 

Setting the `BT_BUF_ACL_TX_SIZE` to 69 when building the `hci_usb` sample will fix this (at least for SMP with LE Secure Connections). 

The problem does not exist for e.g. the `nrf52840dk_nrf52840` as the controller is built together with the `hci_usb` sample, and thus the buffer size reported by the controller (with either Read Buffer Size or Read Buffer Size V2) is the same buffer size the `hci_usb` sample supports, and the host (`native_posix`) can then properly fragment the data. 

Basically we have a 3-build situation here:
Netcore with buffer sizes `X_1` and `Y_1`
Appcore (`hci_usb`) with buffer sizes `X_2` and `Y_2`
native_posix with buffer sizes `X_3` and `Y_3`

where the issue is apparent if `X_1 > X_2` and `X_3 > X_2` and the data length sent is > `X_2`. 

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and flash the controller to the netcore (e.g. `samples/bluetooth/hci_rpmsg`)
2. Compile and flash the `samples/bluetooth/hci_usb` to the appcore
3. Compile and run e.g. the BT shell on the `native_posix` board, and connect to and pair with a device support LE Secure Connections

**Expected behavior**
The expected behavior is that the `hci_usb` can handle any buffer size used by the controller. 

**Impact**
It is just an annoyance, as the buffer sizes used for the sample can simply be increased. 

**Logs and console output**
Can be created on request

**Environment (please complete the following information):**
Controller running on the netcore of the `nrf5340dk` board
`hci_usb` sample running on the appcore of the `nrf5340dk` board
BT shell running on the `native_posix` board on Linux
Commit SHA: a703cbe9ce1315f2460d86de7ee8a55190cf7995

**Additional context**
This problem likely also exists for the `samples/bluetooth/hci_uart`

carlescufi, 2022-08-10:
@Thalley do you plan to address this?

Thalley, 2022-08-10:
Not at this point, no.

closed 2022-11-08 by github-actions[bot]